WPI-AIM/ambf #3: Build Faliure

vidurvij, 2019-04-08:
Getting this error when trying to build project from github clone:

In file included from /media/vidur/DATA/ambf/ambf_ros_modules/ambf_comm/include/ambf_comm/Object.h:49:0,
                 from /media/vidur/DATA/ambf/external/chai3d-bullet/src/CBulletGenericObject.h:56,
                 from /media/vidur/DATA/ambf/ambf_framework/afGripper.h:51,
                 from /media/vidur/DATA/ambf/ambf_framework/afGripper.cpp:46:
/media/vidur/DATA/ambf/ambf_ros_modules/ambf_comm/include/ambf_comm/ObjectRosCom.h:49:10: fatal error: ambf_msgs/ObjectState.h: No such file or directory
 #include "ambf_msgs/ObjectState.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
CMakeFiles/ambf_framework.dir/build.make:62: recipe for target 'CMakeFiles/ambf_framework.dir/ambf_framework/afGripper.cpp.o' failed
make[2]: *** [CMakeFiles/ambf_framework.dir/ambf_framework/afGripper.cpp.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/ambf_framework.dir/all' failed
make[1]: *** [CMakeFiles/ambf_framework.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2

adnanmunawar, 2019-04-08:
Can you please clear your build folder by running `cd ~/ambf/build/ && rm -rf *` and then rerun cmake as `cmake ..` and finally use `make -j8` rather than `make` in the build folder. This should hopefully be a temporary fix while I am still looking to fix it properly.

closed 2019-04-08 by adnanmunawar

vidurvij, 2019-04-08:
I tried the above instruction but the build error persists although at a different completion percentage.

[ 32%] Building C object CMakeFiles/ambf_framework.dir/external/openal/Alc/backends/mmdevapi.c.o
/home/vidur/Desktop/ambf/external/openal/Alc/backends/alsa.c:34:10: fatal error: alsa/asoundlib.h: No such file or directory
 #include <alsa/asoundlib.h>
          ^~~~~~~~~~~~~~~~~~
compilation terminated.
CMakeFiles/ambf_framework.dir/build.make:2918: recipe for target 'CMakeFiles/ambf_framework.dir/external/openal/Alc/backends/alsa.c.o' failed
make[2]: *** [CMakeFiles/ambf_framework.dir/external/openal/Alc/backends/alsa.c.o] Error 1
make[2]: *** Waiting for unfinished jobs....
[ 32%] Building C object CMakeFiles/ambf_framework.dir/external/openal/Alc/backends/null.c.o
CMakeFiles/Makefile2:71: recipe for target 'CMakeFiles/ambf_framework.dir/all' failed
make[1]: *** [CMakeFiles/ambf_framework.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2

adnanmunawar, 2019-04-08:
That is an external library, not part of the AMBF. Try `sudo apt-get install libasound2-dev` and then remake

vidurvij, 2019-04-08:
It worked. Thanks for the help.

cshreyastech, 2019-05-17:
Came across the same issue. Can this be added to the setup/prerequisite process

commit 39734b37 (2019-04-08): Fixed #3 by explicitly adding catkin's targets as build dependency