WPI-AIM/ambf #5: Purpose of the joint message in object command

vidurvij, 2019-04-09:
[https://github.com/WPI-AIM/ambf/blob/master/ambf_ros_modules/ambf_msgs/msg/ObjectCmd.msg](url)

adnanmunawar, 2019-04-10:
We are working on the documentation. Here is a start : 

https://github.com/WPI-AIM/ambf/wiki/Control-AMBF-Simulator-Using-ROS-Messages

adnanmunawar, 2019-04-14:
A added a basic example written in python that leverages the messages the Python client. Here is the link to the example

https://github.com/WPI-AIM/ambf/wiki/A-Basic-Example

vidurvij, 2019-04-22:
Hi Adnan,
I have been trying to access the articles above but it says I dont have the
permission to edit the wiki. And it redirects me to ambf wiki landing page.
Thanks
Vidur

adnanmunawar, 2019-04-22:
Sorry about that. Have been restructuring the wiki a bit so the old link became deprecated. Here is the new link. 
https://github.com/WPI-AIM/ambf/wiki/Understanding-ROS-Message-Payloads

And then to get a hands-on example. Please look at these two:

https://github.com/WPI-AIM/ambf/wiki/Controlling-dVRK-Manipulators

https://github.com/WPI-AIM/ambf/wiki/Controlling-a-Toy-Car

vidurvij, 2019-04-22:
How do we install the module ambf_client used in the example code ?

adnanmunawar, 2019-04-22:
@vidurvij I moved this to a new issue

closed 2019-04-23 by adnanmunawar