WPI-AIM/ambf #8: Importing prespective file for dVRK raises this error in the terminal

vidurvij, 2019-04-22:
GtkDialog mapped without a transient parent. This is discouraged.
PerspectiveManager._create_perspective(ecm-psm-mtm-rqt-gui, False)
MainWindow.save_settings()
PluginManager.save_settings()
PluginManager.save_settings() completed
Traceback (most recent call last):
  File "/home/vidur/miniconda3/lib/python3.7/site-packages/shiboken2/files.dir/shibokensupport/signature/loader.py", line 111, in seterror_argument
    return errorhandler.seterror_argument(args, func_name)
  File "/home/vidur/miniconda3/lib/python3.7/site-packages/shiboken2/files.dir/shibokensupport/signature/errorhandler.py", line 97, in seterror_argument
    update_mapping()
  File "/home/vidur/miniconda3/lib/python3.7/site-packages/shiboken2/files.dir/shibokensupport/signature/mapping.py", line 255, in update
    g.update(g[proc_name]())
  File "/home/vidur/miniconda3/lib/python3.7/site-packages/shiboken2/files.dir/shibokensupport/signature/mapping.py", line 391, in init_QtCore
    "QJsonObject": typing.Dict[str, PySide2.QtCore.QJsonValue],
NameError: name 'PySide2' is not defined
Fatal Python error: seterror_argument did not receive a result

Current thread 0x00007f5815ccb740 (most recent call first):
  File "<string>", line 1 in <module>
  File "/opt/ros/melodic/lib/python2.7/dist-packages/qt_gui/perspective_manager.py", line 410 in _import_value
  File "/opt/ros/melodic/lib/python2.7/dist-packages/qt_gui/perspective_manager.py", line 400 in _convert_values
  File "/opt/ros/melodic/lib/python2.7/dist-packages/qt_gui/perspective_manager.py", line 403 in _convert_values
  File "/opt/ros/melodic/lib/python2.7/dist-packages/qt_gui/perspective_manager.py", line 403 in _convert_values
  File "/opt/ros/melodic/lib/python2.7/dist-packages/qt_gui/perspective_manager.py", line 403 in _convert_values
  File "/opt/ros/melodic/lib/python2.7/dist-packages/qt_gui/perspective_manager.py", line 342 in import_perspective_from_file
  File "/opt/ros/melodic/lib/python2.7/dist-packages/qt_gui/perspective_manager.py", line 329 in _on_import_perspective
  File "/opt/ros/melodic/lib/python2.7/dist-packages/qt_gui/main.py", line 557 in main
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rqt_gui/main.py", line 59 in main
  File "/opt/ros/melodic/lib/rqt_gui/rqt_gui", line 13 in <module>
Aborted (core dumped)

adnanmunawar, 2019-04-22:
This seems to be an issue with RQT install at your side and not related to AMBF. What falvour/version of Linux are you using

adnanmunawar, 2019-05-02:
I cannot reproduce this issue and it does not seem related to ambf. Closing this, feel free to reopen if you have more information

closed 2019-05-02 by adnanmunawar