WPI-AIM/ambf #10: Publish joint names/positions will turn off

nag92, 2019-04-27:
ObjectCmd seems to default the publish the names/positions to "OFF"
so if I send a cmd to the model without always turning the fields to on I stop getting joint
data. 

I think that the default values should be turned on or maybe should be a service request.

adnanmunawar, 2019-04-27:
Good catch. Although I was testing it yesterday and I think the message shouldn’t default back once you stop publishing to it and stay to the last set value. I ll test this again just to be sure

adnanmunawar, 2019-05-01:
So based on the latest tests, the relevant afObject stops publishing the joint names/positions and children names if set to false in the Command message and not automatically. This is intended behavior and hence closing this issue

closed 2019-05-01 by adnanmunawar