WPI-AIM/ambf #11: Build issue

nkalavak, 2019-05-01:
I am running into an error when I use the make command.

```
nivii@nivii-Alienware-Aurora-R7:~/ambf/build$ make
Scanning dependencies of target ambf_framework
[  1%] Building CXX object CMakeFiles/ambf_framework.dir/ambf_framework/afInputDevices.cpp.o
In file included from /home/nivii/ambf/external/chai3d/src/chai3d.h:80:0,
                 from /home/nivii/ambf/external/chai3d-bullet/src/CBulletGenericObject.h:50,
                 from /home/nivii/ambf/external/chai3d-bullet/src/CBulletMultiMesh.h:50,
                 from /home/nivii/ambf/ambf_framework/afSoftMultiMesh.h:52,
                 from /home/nivii/ambf/ambf_framework/afFramework.h:53,
                 from /home/nivii/ambf/ambf_framework/afInputDevices.h:51,
                 from /home/nivii/ambf/ambf_framework/afInputDevices.cpp:46:
/home/nivii/ambf/external/chai3d/src/devices/CdVRKDevices.h:53:26: fatal error: dvrk_arm/Arm.h: No such file or directory
 #include <dvrk_arm/Arm.h>
                          ^
compilation terminated.
CMakeFiles/ambf_framework.dir/build.make:62: recipe for target 'CMakeFiles/ambf_framework.dir/ambf_framework/afInputDevices.cpp.o' failed
make[2]: *** [CMakeFiles/ambf_framework.dir/ambf_framework/afInputDevices.cpp.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/ambf_framework.dir/all' failed
make[1]: *** [CMakeFiles/ambf_framework.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

```

Does the ambf repo require setting up other dvrk related files before working with it?

adnanmunawar, 2019-05-01:
Can you please specify your build system and distro. The `dvrk_arm/Arm.h` is part of the ambf simulator and located in the `ambf_ros_modules` and is built only if ROS install is detected by CMake. And no, you do not require any dVRK dependencies for the simulator

nkalavak, 2019-05-01:
I am using Ubuntu 16.04 with ROS Kinetic.

The output of cmake is:

```
nivii@nivii-Alienware-Aurora-R7:~/ambf/build$ cmake ..
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libGL.so  
CMake Warning at CMakeLists.txt:99 (find_package):
  By not providing "Findcatkin.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "catkin", but
  CMake did not find one.

  Could not find a package configuration file provided by "catkin" with any
  of the following names:

    catkinConfig.cmake
    catkin-config.cmake

  Add the installation prefix of "catkin" to CMAKE_PREFIX_PATH or set
  "catkin_DIR" to a directory containing one of the above files.  If "catkin"
  provides a separate development package or SDK, be sure it has been
  installed.

USINGX86_64_ASM
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   program_options
--   filesystem
--   system
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   program_options
--   filesystem
--   system
-- Configuring done
-- Generating done
-- Build files have been written to: /home/nivii/ambf/build
```

nkalavak, 2019-05-01:
Never mind. I think it was the issue you mentioned wrt CMake detecting ROS install.

closed 2019-05-01 by nkalavak

MitchGaines, 2019-09-30:
For those of you running into this issue, I simply ran the following command in the root directory of ambf prior to running `make`:

`source /opt/ros/<ros version name>/setup.bash`

That seemed to fix it.