WPI-AIM/ambf #15: build failure nvidia TX2 - cpuid.h: No such file or directory

cshreyastech, 2019-05-17:
Trying to build in nvidia TX2 and running into below issue. This seems to be a known issue with TX2 but wanted to have it marked to track.

[ 29%] Building C object CMakeFiles/ambf_framework.dir/external/openal/Alc/helpers.c.o
/home/nvidia/ambf/external/openal/Alc/helpers.c:59:19: fatal error: cpuid.h: No such file or directory
compilation terminated.
CMakeFiles/ambf_framework.dir/build.make:3518: recipe for target 'CMakeFiles/ambf_framework.dir/external/openal/Alc/helpers.c.o' failed
make[2]: *** [CMakeFiles/ambf_framework.dir/external/openal/Alc/helpers.c.o] Error 1
CMakeFiles/Makefile2:711: recipe for target 'CMakeFiles/ambf_framework.dir/all' failed
make[1]: *** [CMakeFiles/ambf_framework.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2

cshreyastech, 2019-05-19:
I have reached out to intel on this issue.
https://github.com/IntelRealSense/librealsense/issues/1501

adnanmunawar, 2019-05-19:
Thanks for opening up this issue. I think we have a TX2 in the lab as well. I shall try to test it on my end as well

adnanmunawar, 2019-09-03:
Fixed with commit 18db3e2aa8276f8caa762a056b631c8d71ae712f

closed 2019-09-03 by adnanmunawar