WPI-AIM/ambf #16: Build Failure

i6zhang, 2019-07-02:
MacBook Pro 2018 13 Inch Mojave 10.14.5

[  0%] Building C object CMakeFiles/ambf_framework.dir/external/glew/src/glew.c.o
/Users/isabelzhang/ambf/external/glew/src/glew.c:1599:1: error: unknown type name 'PFNGLPROGRAMUNIFORM1DEXTPROC'
PFNGLPROGRAMUNIFORM1DEXTPROC __glewProgramUniform1dEXT = NULL;
^
/Users/isabelzhang/ambf/external/glew/src/glew.c:1600:1: error: unknown type name 'PFNGLPROGRAMUNIFORM1DVEXTPROC'
PFNGLPROGRAMUNIFORM1DVEXTPROC __glewProgramUniform1dvEXT = NULL;
^
/Users/isabelzhang/ambf/external/glew/src/glew.c:1607:1: error: unknown type name 'PFNGLPROGRAMUNIFORM2DEXTPROC'
PFNGLPROGRAMUNIFORM2DEXTPROC __glewProgramUniform2dEXT = NULL;
^
/Users/isabelzhang/ambf/external/glew/src/glew.c:1608:1: error: unknown type name 'PFNGLPROGRAMUNIFORM2DVEXTPROC'
PFNGLPROGRAMUNIFORM2DVEXTPROC __glewProgramUniform2dvEXT = NULL;
^
/Users/isabelzhang/ambf/external/glew/src/glew.c:1615:1: error: unknown type name 'PFNGLPROGRAMUNIFORM3DEXTPROC'
PFNGLPROGRAMUNIFORM3DEXTPROC __glewProgramUniform3dEXT = NULL;
^
/Users/isabelzhang/ambf/external/glew/src/glew.c:1616:1: error: unknown type name 'PFNGLPROGRAMUNIFORM3DVEXTPROC'
PFNGLPROGRAMUNIFORM3DVEXTPROC __glewProgramUniform3dvEXT = NULL;
^
/Users/isabelzhang/ambf/external/glew/src/glew.c:1623:1: error: unknown type name 'PFNGLPROGRAMUNIFORM4DEXTPROC'
PFNGLPROGRAMUNIFORM4DEXTPROC __glewProgramUniform4dEXT = NULL;
^
/Users/isabelzhang/ambf/external/glew/src/glew.c:1624:1: error: unknown type name 'PFNGLPROGRAMUNIFORM4DVEXTPROC'
PFNGLPROGRAMUNIFORM4DVEXTPROC __glewProgramUniform4dvEXT = NULL;
^
/Users/isabelzhang/ambf/external/glew/src/glew.c:1631:1: error: unknown type name 'PFNGLPROGRAMUNIFORMMATRIX2DVEXTPROC'; did you mean 'PFNGLPROGRAMUNIFORMMATRIX2FVEXTPROC'?
PFNGLPROGRAMUNIFORMMATRIX2DVEXTPROC __glewProgramUniformMatrix2dvEXT = NULL;
^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
PFNGLPROGRAMUNIFORMMATRIX2FVEXTPROC
/usr/local/include/GL/glew.h:9837:28: note: 'PFNGLPROGRAMUNIFORMMATRIX2FVEXTPROC' declared here
typedef void (GLAPIENTRY * PFNGLPROGRAMUNIFORMMATRIX2FVEXTPROC) (GLuint program, GLint location, GLsizei count, GLboolean transpose, const GLfloat* value);
                           ^
/Users/isabelzhang/ambf/external/glew/src/glew.c:1633:1: error: unknown type name 'PFNGLPROGRAMUNIFORMMATRIX2X3DVEXTPROC'; did you mean 'PFNGLPROGRAMUNIFORMMATRIX2X3FVEXTPROC'?
PFNGLPROGRAMUNIFORMMATRIX2X3DVEXTPROC __glewProgramUniformMatrix2x3dvEXT = NULL;
^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
PFNGLPROGRAMUNIFORMMATRIX2X3FVEXTPROC
/usr/local/include/GL/glew.h:9838:28: note: 'PFNGLPROGRAMUNIFORMMATRIX2X3FVEXTPROC' declared here
typedef void (GLAPIENTRY * PFNGLPROGRAMUNIFORMMATRIX2X3FVEXTPROC) (GLuint program, GLint location, GLsizei count, GLboolean transpose, const GLfloat* value);
                           ^
/Users/isabelzhang/ambf/external/glew/src/glew.c:1635:1: error: unknown type name 'PFNGLPROGRAMUNIFORMMATRIX2X4DVEXTPROC'; did you mean 'PFNGLPROGRAMUNIFORMMATRIX2X4FVEXTPROC'?
PFNGLPROGRAMUNIFORMMATRIX2X4DVEXTPROC __glewProgramUniformMatrix2x4dvEXT = NULL;
^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
PFNGLPROGRAMUNIFORMMATRIX2X4FVEXTPROC
/usr/local/include/GL/glew.h:9839:28: note: 'PFNGLPROGRAMUNIFORMMATRIX2X4FVEXTPROC' declared here
typedef void (GLAPIENTRY * PFNGLPROGRAMUNIFORMMATRIX2X4FVEXTPROC) (GLuint program, GLint location, GLsizei count, GLboolean transpose, const GLfloat* value);
                           ^
/Users/isabelzhang/ambf/external/glew/src/glew.c:1637:1: error: unknown type name 'PFNGLPROGRAMUNIFORMMATRIX3DVEXTPROC'; did you mean 'PFNGLPROGRAMUNIFORMMATRIX3FVEXTPROC'?
PFNGLPROGRAMUNIFORMMATRIX3DVEXTPROC __glewProgramUniformMatrix3dvEXT = NULL;
^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
PFNGLPROGRAMUNIFORMMATRIX3FVEXTPROC
/usr/local/include/GL/glew.h:9840:28: note: 'PFNGLPROGRAMUNIFORMMATRIX3FVEXTPROC' declared here
typedef void (GLAPIENTRY * PFNGLPROGRAMUNIFORMMATRIX3FVEXTPROC) (GLuint program, GLint location, GLsizei count, GLboolean transpose, const GLfloat* value);
                           ^
/Users/isabelzhang/ambf/external/glew/src/glew.c:1639:1: error: unknown type name 'PFNGLPROGRAMUNIFORMMATRIX3X2DVEXTPROC'; did you mean 'PFNGLPROGRAMUNIFORMMATRIX3X2FVEXTPROC'?
PFNGLPROGRAMUNIFORMMATRIX3X2DVEXTPROC __glewProgramUniformMatrix3x2dvEXT = NULL;
^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
PFNGLPROGRAMUNIFORMMATRIX3X2FVEXTPROC
/usr/local/include/GL/glew.h:9841:28: note: 'PFNGLPROGRAMUNIFORMMATRIX3X2FVEXTPROC' declared here
typedef void (GLAPIENTRY * PFNGLPROGRAMUNIFORMMATRIX3X2FVEXTPROC) (GLuint program, GLint location, GLsizei count, GLboolean transpose, const GLfloat* value);
                           ^
/Users/isabelzhang/ambf/external/glew/src/glew.c:1641:1: error: unknown type name 'PFNGLPROGRAMUNIFORMMATRIX3X4DVEXTPROC'; did you mean 'PFNGLPROGRAMUNIFORMMATRIX3X4FVEXTPROC'?
PFNGLPROGRAMUNIFORMMATRIX3X4DVEXTPROC __glewProgramUniformMatrix3x4dvEXT = NULL;
^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
PFNGLPROGRAMUNIFORMMATRIX3X4FVEXTPROC
/usr/local/include/GL/glew.h:9842:28: note: 'PFNGLPROGRAMUNIFORMMATRIX3X4FVEXTPROC' declared here
typedef void (GLAPIENTRY * PFNGLPROGRAMUNIFORMMATRIX3X4FVEXTPROC) (GLuint program, GLint location, GLsizei count, GLboolean transpose, const GLfloat* value);
                           ^
/Users/isabelzhang/ambf/external/glew/src/glew.c:1643:1: error: unknown type name 'PFNGLPROGRAMUNIFORMMATRIX4DVEXTPROC'; did you mean 'PFNGLPROGRAMUNIFORMMATRIX4FVEXTPROC'?
PFNGLPROGRAMUNIFORMMATRIX4DVEXTPROC __glewProgramUniformMatrix4dvEXT = NULL;
^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
PFNGLPROGRAMUNIFORMMATRIX4FVEXTPROC
/usr/local/include/GL/glew.h:9843:28: note: 'PFNGLPROGRAMUNIFORMMATRIX4FVEXTPROC' declared here
typedef void (GLAPIENTRY * PFNGLPROGRAMUNIFORMMATRIX4FVEXTPROC) (GLuint program, GLint location, GLsizei count, GLboolean transpose, const GLfloat* value);
                           ^
/Users/isabelzhang/ambf/external/glew/src/glew.c:1645:1: error: unknown type name 'PFNGLPROGRAMUNIFORMMATRIX4X2DVEXTPROC'; did you mean 'PFNGLPROGRAMUNIFORMMATRIX4X2FVEXTPROC'?
PFNGLPROGRAMUNIFORMMATRIX4X2DVEXTPROC __glewProgramUniformMatrix4x2dvEXT = NULL;
^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
PFNGLPROGRAMUNIFORMMATRIX4X2FVEXTPROC
/usr/local/include/GL/glew.h:9844:28: note: 'PFNGLPROGRAMUNIFORMMATRIX4X2FVEXTPROC' declared here
typedef void (GLAPIENTRY * PFNGLPROGRAMUNIFORMMATRIX4X2FVEXTPROC) (GLuint program, GLint location, GLsizei count, GLboolean transpose, const GLfloat* value);
                           ^
/Users/isabelzhang/ambf/external/glew/src/glew.c:1647:1: error: unknown type name 'PFNGLPROGRAMUNIFORMMATRIX4X3DVEXTPROC'; did you mean 'PFNGLPROGRAMUNIFORMMATRIX4X3FVEXTPROC'?
PFNGLPROGRAMUNIFORMMATRIX4X3DVEXTPROC __glewProgramUniformMatrix4x3dvEXT = NULL;
^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
PFNGLPROGRAMUNIFORMMATRIX4X3FVEXTPROC
/usr/local/include/GL/glew.h:9845:28: note: 'PFNGLPROGRAMUNIFORMMATRIX4X3FVEXTPROC' declared here
typedef void (GLAPIENTRY * PFNGLPROGRAMUNIFORMMATRIX4X3FVEXTPROC) (GLuint program, GLint location, GLsizei count, GLboolean transpose, const GLfloat* value);
                           ^
/Users/isabelzhang/ambf/external/glew/src/glew.c:1996:1: error: unknown type name 'PFNGLGETPOINTERVPROC'
PFNGLGETPOINTERVPROC __glewGetPointerv = NULL;
^
/Users/isabelzhang/ambf/external/glew/src/glew.c:2448:1: error: unknown type name 'PFNGLCLEARDEPTHFOESPROC'; did you mean 'PFNGLCLEARDEPTHFPROC'?
PFNGLCLEARDEPTHFOESPROC __glewClearDepthfOES = NULL;
^~~~~~~~~~~~~~~~~~~~~~~
PFNGLCLEARDEPTHFPROC
/usr/local/include/GL/glew.h:3821:28: note: 'PFNGLCLEARDEPTHFPROC' declared here
typedef void (GLAPIENTRY * PFNGLCLEARDEPTHFPROC) (GLclampf d);
                           ^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
20 errors generated.
make[2]: *** [CMakeFiles/ambf_framework.dir/external/glew/src/glew.c.o] Error 1
make[1]: *** [CMakeFiles/ambf_framework.dir/all] Error 2
make: *** [all] Error 2

adnanmunawar, 2019-07-03:
Let's try to use a separate install of glew. Can you clone glew from here.

https://github.com/nigels-com/glew 

Then as the instruction says
```
cd glew/auto
make
```
Then 
```
cd ../build/
cmake cmake/.
make -j8
sudo make install
```
Then, please try replacing the ambf/CMakeLists.txt file with this one and then clean build folder (remove it) and rerun cmake and make.

[CMakeLists.txt](https://github.com/WPI-AIM/ambf/files/3355392/CMakeLists.txt)

i6zhang, 2019-07-11:
I tested this solution and it does allow me to build the application.

closed 2019-09-13 by adnanmunawar