WPI-AIM/ambf #29: Feature enhancement: personalize message type for topics

cshreyastech, 2019-09-10:
All all topics have same message types which could have irrelevant message types. 

For example light topics has torque and velocity which are irrelevant to the topic.
Creating personalized message types for each topic category will make the topics much realistic and modular.

As suggested by @adnanmunawar , customizing light and camera topics would be good start.

@adnanmunawar, can you also list out other topic topic categories and their corresponding relevant messages to be published?

Thanks

adnanmunawar, 2019-09-13:
Thanks for the enhancement request Shreyas. So let's see, how about we start to implement something along these lines for the camera:

Looking at file https://github.com/WPI-AIM/ambf/blob/master/ambf_models/descriptions/world/world.yaml#L32

``` yaml
camera1:
  name: default_camera
  location: {x: 4.0, y: 0.0, z: 2.0}
  look at: {x: 0.0, y: 0.0, z: -0.5}
  up: {x: 0.0, y: 0.0, z: 1.0}
  clipping plane: {near: 0.01, far: 10.0}
  field view angle: 0.8
  monitor: 0
```
The useful fields are location, look at and filed view angle. We can definitely add more in the future.

Now looking at the light, in the file: https://github.com/WPI-AIM/ambf/blob/master/ambf_models/descriptions/world/world.yaml#L6

``` yaml
light1:
  namespace: /ambf/env/lights/
  name: light_left
  location: {x: 0.0, y: 0.5, z: 2.5}
  direction: {x: 0, y: 0, z: -1.0}
  spot exponent: 0.3
  shadow quality: 0
  cutoff angle: 0.7
```
All the fields except the namespace, name and shadow quality seem like they should be dynamically controllable.

cshreyastech, 2020-03-26:
yes, that sounds good. I can work on it.

adnanmunawar, 2020-11-19:
ambf-1.0 has introduced separate messages for each type of object (rigid bodies, cameras, sensors, actuators, lights etc)

closed 2020-11-19 by adnanmunawar