WPI-AIM/ambf #32: Python client set_pos setting up all joints

prajankya, 2019-09-13:
I was using python client to set position for Kuka robot (```./ambf_simulator -l 8``` i.e. ```blender-kuka.yaml```)

```python
#! /usr/bin/env python

import time

from ambf_client import Client

_client = Client()

_client.connect()

print('\n\n----')
print("List of Objects")
print(_client.get_obj_names())

kuka_handle = _client.get_obj_handle('base')

time.sleep(0.2)
print('\n\n----')

print('Kuka Base Pos:')
print(kuka_handle.get_pos())

print('\n\n----')

print('Number of Joints in kuka:')
print(kuka_handle.get_num_joints())

print('Name of kuka\'s children:')
print(kuka_handle.get_joint_names())

print(kuka_handle.get_name())
print(kuka_handle.get_pos_command())
print(kuka_handle.get_pose())

kuka_handle.set_joint_pos(0, 1.57)
# kuka_handle.set_joint_effort(0, 10)
# kuka_handle.set_joint_effort(1, 10)

print('\n\n----')
raw_input("Let's clean up. Press Enter to continue...")
# Lastly to cleanup
_client.clean_up()
```

Also the blender-kuka.yaml file was updated to change PID values for all joints as a work around for the issue.
blender-kuka.yaml
```yaml
# AMBF Version: (0.1)
# Generated By: ambf_addon for Blender (2.79.0)
# Link: https://github.com/WPI-AIM/ambf_addon
# Generated on: 2019-04-01 17:46:58
bodies: [BODY base, BODY link1, BODY link2, BODY link3, BODY link4, BODY link5, BODY
    link6, BODY link7]
joints: [JOINT base-link1, JOINT link1-link2, JOINT link2-link3, JOINT link3-link4,
  JOINT link4-link5, JOINT link5-link6, JOINT link6-link7]
high resolution path: ../../../meshes/blender_afmb/kuka_lwr/high_res/
low resolution path: ../../../meshes/blender_afmb/kuka_lwr/low_res/
ignore inter-collision: true
namespace: /ambf/env/
joint erp: 0.5
joint cfm: 0.3
BODY base:
  name: base
  mesh: base.STL
  mass: 0.0
  collision margin: 0.001
  scale: 1.0
[... 296 lines cut ...]
    x: 0.0
    y: 0.0
    z: 0.0
  orientation: 
    x: 0.0
    y: 0.0
    z: 0.0
    w: 0.0
wrench: 
  force: 
    x: 0.0
    y: 0.0
    z: 0.0
  torque: 
    x: 0.0
    y: 0.0
    z: 0.0
joint_cmds: [1.5700000524520874, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
position_controller_mask: [True, True, True, True, True, True, True]
publish_children_names: False
publish_joint_names: False
publish_joint_positions: False
```

Ambf simulator is being sent all the joint positions from ambf_client, which as discussed with @adnanmunawar is not how it is should be.

adnanmunawar, 2019-09-13:
Thanks for creating the issue. While trying to fix the overriding of joint cmds in 5174fae4e5d5d603da18d1936aae153c47f8bedb we accidentally introduced this issue

adnanmunawar, 2019-09-13:
This line:
 https://github.com/WPI-AIM/ambf/blob/03e2fa665eb79552f57a5c83530ce6433425eb38/ambf_ros_modules/ambf_comm/scripts/ambf_object.py#L333

Needs to be changed from

```
            self._cmd.position_controller_mask = [1]*n_jnts
```

To
```
            self._cmd.position_controller_mask = [0]*n_jnts
```

@nag92

commit 84c057bb (2019-09-13): Fixed #32
closed 2019-09-13 by adnanmunawar