WPI-AIM/ambf #34: build error

johnathanchiu, 2019-09-20:
Hi, I'm getting this error while running make in the build dir.

```
[  0%] Building C object CMakeFiles/ambf_framework.dir/external/giflib/src/dev2gif.c.o
ambf/external/giflib/src/dev2gif.c:324:47: error: use of undeclared identifier 'FALSE'
    if ((GifFile = EGifOpenFileName(FileName, FALSE)) == NULL ||
                                              ^
ambf/external/giflib/src/dev2gif.c:327:65: error: use of undeclared identifier 'FALSE'
        EGifPutImageDesc(GifFile, 0, 0, ScreenXMax, ScreenYMax, FALSE,
                                                                ^
ambf/external/giflib/src/dev2gif.c:347:30: error: too few arguments to function call, expected 2, have
      1
    if (EGifCloseFile(GifFile) == GIF_ERROR) {
        ~~~~~~~~~~~~~        ^
/usr/local/include/gif_lib.h:132:1: note: 'EGifCloseFile' declared here
int EGifCloseFile(GifFileType *GifFile, int *ErrorCode);
^
ambf/external/giflib/src/dev2gif.c:502:13: warning: implicit declaration of function 'GifLastError' is
      invalid in C99 [-Wimplicit-function-declaration]
    int i = GifLastError();
            ^
ambf/external/giflib/src/dev2gif.c:504:30: error: too few arguments to function call, expected 2, have
      1
    if (EGifCloseFile(GifFile) == GIF_ERROR) {
        ~~~~~~~~~~~~~        ^
/usr/local/include/gif_lib.h:132:1: note: 'EGifCloseFile' declared here
int EGifCloseFile(GifFileType *GifFile, int *ErrorCode);
^
1 warning and 4 errors generated.
make[2]: *** [CMakeFiles/ambf_framework.dir/external/giflib/src/dev2gif.c.o] Error 1
make[1]: *** [CMakeFiles/ambf_framework.dir/all] Error 2
make: *** [all] Error 2
```

adnanmunawar, 2019-09-20:
Can you include your OS and Hardware Specs including CPU Arch (x86x64 or ARM)

johnathanchiu, 2019-09-20:
MacOS Mojave 10.14.6, intel i7 64 bit

adnanmunawar, 2019-09-20:
Thanks, can you also specify the compiler being used and whether or not you have installed Mac OS's **command line tools**

adnanmunawar, 2019-09-21:
It appears that you have GIFLIB (a different version) installed separately on your system as well ( with header files in /usr/local/include/gif_lib.h). These headers are being confused by CMake as the src files in the 'external' folder for giflib are for an older version. Can you also specify the compiler being used and whether or not you have installed Mac OS's command-line tools?

johnathanchiu, 2019-09-22:
I am using apple clang version v11.0.0 and yes, I have the command-line tools installed.

adnanmunawar, 2019-09-23:
I have not been able to reproduce the issue on similar conditions, therefore, while I still try to reproduce this, let's dissect this a bit further

1. Can you please elaborate how you installed `libgif` on your mac, the usual way is to install via homebrew which doesn't place the headers in `/usr/local/include`. Please correct me if I am mistaken. I shall try to install `libgif` the same way as you to see if I can reproduce the issue. Manually copying the files in system folders isn't giving me any issues at the moment.

2. In your local clone, can you modify the root `CMakeLists.txt` file at this line:

https://github.com/WPI-AIM/ambf/blob/master/CMakeLists.txt#L228

```
include_directories (${PROJECT_SOURCE_DIR}/external/giflib/include)
```
to:

```
include_directories (BEFORE ${PROJECT_SOURCE_DIR}/external/giflib/include)
```

then clean your build folder, `rm -rf ~/ambf/build`, recreate the build folder and trying to build again.

adnanmunawar, 2019-09-26:
I installed MacOS Catalina today and now I am able to reproduce the issue. Per the last comment, using the BEFORE keyword in CMake fixed this issue. I found other libraries (jpeg and glew) behave the same way in the new MacOS update. Commit cf7b029d423cb4918aab0713232422c929c4603c seems to fix this.

johnathanchiu, 2019-09-28:
Hi that fixed the initial build error but now I am getting a new one.

`ambf/external/theoraplayer/external/theora/lib/encode.c:1511:10: error: use of undeclared identifier
      'TH_ENCCTL_SET_COMPAT_CONFIG'
    case TH_ENCCTL_SET_COMPAT_CONFIG:{`

To clarify, I am pretty certain I have all the dependencies. I installed everything using homebrew.

adnanmunawar, 2019-09-28:
Thanks for trying that out. Yup, even though I can't see the full error message, I am almost certain that this too is a similar problem as the "therora/lib/encode.c" file contains some identifier that the compiler can't find. To be clear, I think this issue is coming from some external dependencies already installed in your system separately which happen to be different versions and thereby have different header files. Please test out the newest commit to see if it resolves the issue.

adnanmunawar, 2019-10-11:
Is there any update on this issue? Did you get a chance to try the fixes.

johnathanchiu, 2019-10-12:
Yes! I was just able to build!

closed 2019-10-12 by johnathanchiu