WPI-AIM/ambf #41: Topic definition and namespace handling for multiple robots for controller design

DhruvKoolRajamani, 2019-12-07:
Currently, spawning multiple instances of the same model appends a number to the link like this:  

[image]

My suggestion is to append the number to the namespace:

[image]

And also have a feature of maybe storing the namespaces in a list on the param server if ros is running OR publish a topic with the namespace prefixes.

[image]

The need for this is for designing a controller, modifying the prefix as compared to the link names might be cleaner when describing the details of a robot in one or many environments.

adnanmunawar, 2020-02-06:
Thanks for creating this Dhruv. As per our discussion earlier, the design that I had in mind was to use a logic similar to directory management. Therefore a newly added object with a colliding name is resolved by appending an index with it's name. One could argue that a namespace modification, as is employed by Gazebo, is another viable solution. And I agree.

So essentially, this boils down to how you view the simulation and the associated objects. In Blender such naming collision is handled by renaming with an index, somewhat similar to AMBF although not identical. So AMBF is not alone is taking this approach.  

That being said, I do think we should have an implementation wherein we provide a flag that allows users to chose what kind of naming resolution they prefer. However, at the moment, there are a few other important updates that I need to attend to before I delve into that.

DhruvKoolRajamani, 2020-02-07:
Thanks for the clarification! I definitely agree with your logic, ambf is a platform which leverages Blender for meshes and it would make sense to propagate a similar naming convention. The feature to allow users to switch between different conventions seems neat as well and I'll be happy to assist you as and when the issue comes along your agenda.

closed 2020-04-08 by DhruvKoolRajamani