WPI-AIM/ambf #42: python client issue: getting object of world makes the code crash

VigneshManoj, 2020-02-15:
In the make function of '/ambf/ambf_ros_modules/ambf_comm/scripts/ambf_env.py' script the world handle is assigned using the function get_obj_handle (present in '/ambf/ambf_ros_modules/ambf_comm/scripts/ambf_client.py'). But since the world doesn't have children, it is unable to set the set_publish_children_names_flag true.
Need to add an if case which excludes running that for World object.

closed 2020-02-16 by adnanmunawar

adnanmunawar, 2020-02-16:
Checkout the new puzzle environment I just added. This works with the original test code.

Please note in https://github.com/WPI-AIM/ambf/blob/master/ambf_ros_modules/ambf_comm/scripts/ambf_env.py#L87

that an object named "PegBase" is hard coded in the ENV code. Since this was just a test implementation and I wanted to make the Torus automatically go into a PegBase, I explicity created this line in there. Using this line, I was defining a reward function. 

This probably is of no use to you, so you can remove the pertaining code. But its a good place to start with the new environment I just created. You can run it as

```
./ambf_simulator -a <path_to_ambf>/ambf_models/descriptions/multi-bodies/puzzles/peg_and_base.yaml
```
Then run the Python test file:
https://github.com/WPI-AIM/ambf/blob/master/ambf_ros_modules/ambf_comm/scripts/tests/rl_test.py

VigneshManoj, 2020-02-16:
Ok thank you!
Yea I had noticed the peg base and realized you were using it for something
else. I shall check it out and see.

commit 7f83705a (2020-02-16): Fixes #42