WPI-AIM/ambf #50: Can't run python client

nepacheco, 2020-02-25:
Installed ambf and ros. Tried to run the python client, but this error pops up instead. 
System: Ubuntu 18.04, ROS Melodic, Python 2.7

``` 
 File "/home/nicholas/ambf/build/devel/lib/python2.7/dist-packages/ambf_client/__init__.py", line 35, in <module>
    exec(__fh.read())
  File "<string>", line 49, in <module>
  File "/home/nicholas/ambf/ambf_ros_modules/ambf_comm/scripts/ambf_object.py", line 49, in <module>
    from external.tf_function.transformations import quaternion_from_euler, euler_from_quaternion
ImportError: No module named external.tf_function.transformations
```

adnanmunawar, 2020-02-25:
Thanks for creating the issue. This is a result of the recent PR by a collaborator. He has issued a fix which I haven't go over yet, but meanwhile just roll back on commit on the master brank.
I.e. in your ambf folder, run:

```
git checkout b7bd0ca060381f2c6bad2b81f329005393248fbe
```

And then when I am able to merge the PR, you can update to the latest commit

adnanmunawar, 2020-02-25:
I just merged the PR which seems to fix the issue. Please give it a try but checking out the latest commit

closed 2020-02-28 by adnanmunawar

jdalmeida, 2020-03-10:
This worked thank you!