WPI-AIM/ambf #54: [Release] Allow multiple rosmasters using ports

DhruvKoolRajamani, 2020-03-25:
Refer to the [thread](https://answers.ros.org/question/193062/how-to-run-multiple-independent-gazebo-instances-on-the-same-machine/) to setup multiple rosmasters.

- [ ] Allow -p port to be specified for ambf client instances.
- [ ] Run profiling to test the performance of multiple bulletphysics backends being spawned
- [ ] Allow AMBF Client to share backend physics over a UDP port to be accessed by a remote AMBF Graphics client for remote connections. ([Shared Memory](https://github.com/bulletphysics/bullet3/tree/master/examples/SharedMemory))

adnanmunawar, 2020-03-27:
I tested running multiple `roscore`s on the same machine and multiple instances of AMBF and they work as expected.

Usage 

```
Terminal 1
roscore -p <first_port_num>

Terminal 2
export ROS_MASTER_URI=http://localhost:<first_port_number>
./ambf_simulator <any necessary CLIs>

Terminal 3
roscore -p <second_port_num>

Terminal 4
export ROS_MASTER_URI=http://localhost:<second_port_number>
./ambf_simulator <any necessary CLIs>

and so on ...
```

**NOTE:** Make sure to `export` the correct `ROS_MASTER_URI`'s on all the terminals running the Python AMBF client to connect to the proper `roscore` and AMBF instance.

adnanmunawar, 2020-03-30:
This commit f9db7770404e56810c7caff84c33c6917284dac9 allows a user to run ambf with the CLI arg `--ns` and specify a forced prefix to the namespace. 
**Usage:**
```
./ambf_simulator <desired_adf_files> --ns /test/
```
This will result in the string `"/test/"` to be appended to all the topics regardless of whether or not they  are specified in the relevant ADF files with an absolute namespace (e.g. `"/ambf/env/"`).

DhruvKoolRajamani, 2020-04-08:
Can we close this?

closed 2020-04-08 by DhruvKoolRajamani