WPI-AIM/ambf #59: Unable to Build Ubuntu 18.04 and OSX 15.4

pacarvalho, 2020-03-30:
Following the instructions for building outlined in the README I get the following error in both Linux and Unix:

LInux:
```
root@a89495270a39:/home/ambf/build# cmake ..
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error at CMakeLists.txt:92 (find_package):
  By not providing "Findyaml-cpp.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "yaml-cpp",
  but CMake did not find one.

  Could not find a package configuration file provided by "yaml-cpp" with any
  of the following names:

    yaml-cppConfig.cmake
    yaml-cpp-config.cmake

  Add the installation prefix of "yaml-cpp" to CMAKE_PREFIX_PATH or set
  "yaml-cpp_DIR" to a directory containing one of the above files.  If
  "yaml-cpp" provides a separate development package or SDK, be sure it has
  been installed.

-- Configuring incomplete, errors occurred!
See also "/home/ambf/build/CMakeFiles/CMakeOutput.log".
root@a89495270a39:/home/ambf/build# 
```

Unix: 
```
 build git:(master) cmake ..          
CMake Error at CMakeLists.txt:92 (find_package):
  By not providing "Findyaml-cpp.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "yaml-cpp",
  but CMake did not find one.

  Could not find a package configuration file provided by "yaml-cpp" with any
  of the following names:

    yaml-cppConfig.cmake
    yaml-cpp-config.cmake

  Add the installation prefix of "yaml-cpp" to CMAKE_PREFIX_PATH or set
  "yaml-cpp_DIR" to a directory containing one of the above files.  If
  "yaml-cpp" provides a separate development package or SDK, be sure it has
  been installed.

-- Configuring incomplete, errors occurred!
See also "/Users/admin/Software/ambf/build/CMakeFiles/CMakeOutput.log".
```

Before I start trying to debug: Has anyone seen this error before?

pacarvalho, 2020-03-30:
Ah! There were some missing dependencies. Will list them all here based on a barebones Ubuntu:18.04 image.

adnanmunawar, 2020-03-30:
Interesting, I always assumed **yaml-cpp** ships with Ubuntu 14, 16 and 18 but maybe not with the barebones copy. You can install it via

```
sudo apt install libyaml-cpp-dev
```
However, there might be other missing libs such as `Boost` etc. 

On a side note, I would not recommend installing on Mac-OS at the moment.

pacarvalho, 2020-03-30:
The only two missing dependencies were: **libyaml-cpp-dev** and **libboost-all-dev**. Upon installing them cmake succeeded.

closed 2020-03-30 by pacarvalho