WPI-AIM/ambf #60: Make (Build) Issue "fatal error: dvrk_arm/Arm.h: No such file or directory"

pacarvalho, 2020-03-30:
The error reported in #11 appears to occur when attempting to install this package on Ubuntu 18.04 **without** ROS installed. 

Output from `CMake`

```
root@63a651139a01:/ambf/build# cmake ..
-- Boost version: 1.65.1
CMake Warning at CMakeLists.txt:104 (find_package):
  By not providing "Findcatkin.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "catkin", but
  CMake did not find one.

  Could not find a package configuration file provided by "catkin" with any
  of the following names:

    catkinConfig.cmake
    catkin-config.cmake

  Add the installation prefix of "catkin" to CMAKE_PREFIX_PATH or set
  "catkin_DIR" to a directory containing one of the above files.  If "catkin"
  provides a separate development package or SDK, be sure it has been
  installed.

USINGX86_64_ASM
-- Boost version: 1.65.1
-- Found the following Boost libraries:
--   program_options
--   filesystem
--   system
CMake Warning at CMakeLists.txt:359 (find_package):
  By not providing "Findcatkin.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "catkin", but
  CMake did not find one.

  Could not find a package configuration file provided by "catkin" with any
  of the following names:

    catkinConfig.cmake
    catkin-config.cmake

  Add the installation prefix of "catkin" to CMAKE_PREFIX_PATH or set
  "catkin_DIR" to a directory containing one of the above files.  If "catkin"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring done
-- Generating done
-- Build files have been written to: /ambf/build
```

From the CMakeLists.txt definition (shown below) I thought it would not be required. Is ROS a required dependency? 

https://github.com/WPI-AIM/ambf/blob/566e1847714b14ce404c21c2b681e09aa6a72b5e/CMakeLists.txt#L103-L164

I added the line `message(${catkin_FOUND})` to the CMakeLists.txt and it correctly prints out 0 (Not detected).

pacarvalho, 2020-03-30:
Full error message from running `make`

```
Scanning dependencies of target ambf_framework
[  0%] Building CXX object CMakeFiles/ambf_framework.dir/ambf_framework/afFramework.cpp.o
In file included from /ambf/external/chai3d/src/chai3d.h:80:0,
                 from /ambf/external/chai3d-bullet/src/CBulletGenericObject.h:50,
                 from /ambf/external/chai3d-bullet/src/CBulletMultiMesh.h:50,
                 from /ambf/ambf_framework/afSoftMultiMesh.h:52,
                 from /ambf/ambf_framework/afFramework.h:53,
                 from /ambf/ambf_framework/afFramework.cpp:48:
/ambf/external/chai3d/src/devices/CdVRKDevices.h:53:10: fatal error: dvrk_arm/Arm.h: No such file or directory
 #include <dvrk_arm/Arm.h>
          ^~~~~~~~~~~~~~~~
compilation terminated.
CMakeFiles/ambf_framework.dir/build.make:62: recipe for target 'CMakeFiles/ambf_framework.dir/ambf_framework/afFramework.cpp.o' failed
make[2]: *** [CMakeFiles/ambf_framework.dir/ambf_framework/afFramework.cpp.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/ambf_framework.dir/all' failed
make[1]: *** [CMakeFiles/ambf_framework.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
```

adnanmunawar, 2020-03-30:
On Linux it is expected that you have ROS installed. That is where the error is stemming from.  Please install ROS from here. 

http://wiki.ros.org/ROS/Installation

I also think that when you have ROS installed, it should fix the missing `Boost` and `Yaml-cpp` from the issue #59.

We also never explicitly stated that one must install ROS in the README but it should be there for new users. @DhruvKoolRajamani @nag92 . Can some add this in the README and WIKI please?

nag92, 2020-03-30:
sure I will take of that tonight

pacarvalho, 2020-03-30:
I will start from a blank distro again and see it it removes the need to install the dependencies of #59.

adnanmunawar, 2020-03-30:
You may also try removing them from apt-get instead of having to reinstall a clean disk image

nag92, 2020-03-30:
> You may also try removing them from apt-get instead of having to reinstall a clean disk image
Testing in a VM would be easy since you can nuke. There is always the possibility of lingering dependencies.

DhruvKoolRajamani, 2020-03-30:
@pacarvalho you can use these setup scripts to install all the dependencies
https://github.com/FaridTavakol/AIM_Lab_server
setup1 will install ros repositories, setup 2 initializes ROS and build ambf

pacarvalho, 2020-03-30:
@DhruvKoolRajamani Thank you. 

I switched to Docker ROS once @adnanmunawar confirmed Docker was required ;) 

@DhruvKoolRajamani, I am trying to figure out what is the minimal image that runs AMBF. Which version of ROS is required? Running only the core (image `ros:melodic-ros-core-bionic`) results in the following missing dependency:

```
Could NOT find tf (missing: tf_DIR)
-- Could not find the required component 'tf'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "tf" with any of
  the following names:

    tfConfig.cmake
    tf-config.cmake

  Add the installation prefix of "tf" to CMAKE_PREFIX_PATH or set "tf_DIR" to
  a directory containing one of the above files.  If "tf" provides a separate
  development package or SDK, be sure it has been installed.
Call Stack (most recent call first):
  ambf_ros_modules/ambf_comm/CMakeLists.txt:52 (find_package)

-- Configuring incomplete, errors occurred!
```

@adnanmunawar I confirmed that even using Docker ROS `libyaml-cpp-dev` still needs to be installed.

pacarvalho, 2020-03-30:
@FaridTavakol Seems to use the full version but that is a 1GB+ install which would not be ideal for a CICD for instance. 

https://github.com/FaridTavakol/AIM_Lab_server/blob/c684b785075becc1e89f716681e7d612b46f4259/setup1.sh#L19

DhruvKoolRajamani, 2020-03-30:
Ideally you would install `ros-melodic-desktop-full`

I just went through the Docker images from the official ROS repos and it's interesting that they don't have support for the desktop-full tag. While setting up the vm yesterday, we didn't come across the `libyaml-cpp-dev` issue, however, since the future plan is to implement this installation on just ros-base, I will take note of all these dependency issues and update the installation/documentation/CMakeLists file respectively.

As for the `apt-get install` you can perform on top of the ros-base image, you can check out this tree here:
[ros-melodic-desktop](https://aur.archlinux.org/packages/ros-melodic-desktop/)

I believe this should be enough for ambf to work. However, I haven't sifted through the repositories which I will do in the future.

DhruvKoolRajamani, 2020-03-30:
> @FaridTavakol Seems to use the full version but that is a 1GB+ install which would not be ideal for a CICD for instance.
> 
> https://github.com/FaridTavakol/AIM_Lab_server/blob/c684b785075becc1e89f716681e7d612b46f4259/setup1.sh#L19

Yes, you're absolutely right, when setting up the CI we will sift through all the dependencies and keep the necessary ones. (Will look into this in a few days)

pacarvalho, 2020-03-31:
As @DhruvKoolRajamani pointed out it is possible to install AMBF in Docker starting from `ros:melodic-ros-base` and then performing an `apt-get install ros-melodic-desktop-full` to extend the base install. 

Installing `libyaml-cpp-dev` was still required and should probably be added to the dependency list for AMBF.

closed 2020-03-31 by pacarvalho

DhruvKoolRajamani, 2020-03-31:
Hey @pacarvalho that's great, could you also share/setup a PR with the DOCKERFILE? That would be quite helpful to setup a CICD.

pacarvalho, 2020-03-31:
Will do. However, let's have our meeting first so that I know what it is you guys are looking for.