WPI-AIM/ambf #61: a is bring striped out

nag92, 2020-03-30:
I pulled the recent branch and I am getting a weird error causing all the "a"s to be stripped out

```
['/mbf/env/Rightthigh/Stte',
 '/mbf/env/Leftthigh/Stte',
 '/mbf/env/Rightfoot/Stte',
 '/mbf/env/Leftfoot/Stte',
 '/mbf/env/Rightshnk/Stte',
 '/mbf/env/lights/light_left/Stte',
 '/mbf/env/Hip/Stte',
 '/mbf/env/Leftshnk/Stte',
 '/cmers/defult_cmer/Stte',
 '/lights/light_right/Stte']

```

nag92, 2020-03-31:
```
In [7]: from ambf_client import Client

In [8]: _client = Client()

In [9]: _client.connect()
('Found Common Object Namespace as: ', 'a')

In [11]: _client.get_obj_names()
Out[11]: 
['/mbf/env/Rightthigh/Stte',
 '/mbf/env/Leftthigh/Stte',
 '/mbf/env/Rightfoot/Stte',
 '/mbf/env/Leftfoot/Stte',
 '/mbf/env/Rightshnk/Stte',
 '/mbf/env/lights/light_left/Stte',
 '/mbf/env/Hip/Stte',
 '/mbf/env/Leftshnk/Stte',
 '/cmers/defult_cmer/Stte',
 '/lights/light_right/Stte']

```

adnanmunawar, 2020-03-31:
Can you test this with the following commit faa19aab5756277b45c1dddaabbdf68bd77d2db9

nag92, 2020-03-31:
that seems to have worked. What the hell could of caused its strange error?

closed 2020-03-31 by nag92