WPI-AIM/ambf #62: Running ./ambf_simulator Fails w/ "failed initialization" Error on Docker

pacarvalho, 2020-03-31:
It is possible to install AMBF using the Dockerfile below:

```Dockerfile
## Base image
FROM ros:melodic-ros-base

## Set work directory
RUN mkdir /usr/src/ambf
WORKDIR /usr/src/ambf

## Install required packages
# RUN apt-get install -y git cmake
RUN apt-get update && \
    apt-get install -y ros-melodic-desktop-full \
    libasound2-dev libgl1-mesa-dev xorg-dev libyaml-cpp-dev && \
    rm -r /var/lib/apt/lists

## Install AMBF
# Copy files into docker 
COPY . .
# Update the submodules required by ambf
RUN git submodule update --init --recursive
# Create a build subdirectory, cd into it and build
RUN . /opt/ros/melodic/setup.sh && \
    mkdir build && \
    cd build && \
    cmake .. && \
    make

## Entrypoint for the image (command that will execute when invoking docker run)
CMD ["bash"]
```

However, trying to run the resulting installation yields the following message. Is there any place where I can find a log to help me debug? 

```
root@a88d8c19cf92:/usr/src/ambf/bin/lin-x86_64# ./ambf_simulator

____________________________________________________________

ASYNCHRONOUS MULTI-BODY FRAMEWORK SIMULATOR (AMBF Simulator)

                (www.aimlab.wpi.edu)
                  (Copyright 2019)
____________________________________________________________

STARTUP COMMAND LINE OPTIONS: 

InputDevices Application Usage:
  -h [ --help ]                     Show help
  -n [ --ndevs ] arg                Number of Haptic Devices to Load
  -i [ --load_devices ] arg         Index number of devices to load which is 
                                    specified in input_device.yaml
  -e [ --enableforces ] arg         Enable Force Feedback on Haptic Devices
  -p [ --phx_frequency ] arg        Physics Update Frequency (default: 1000 Hz)
  -d [ --htx_frequency ] arg        Haptics Update Frequency (default: 1000 Hz)
  -t [ --fixed_phx_timestep ] arg   Use Fixed Time-Step for Physics (default: 
                                    False)
  -f [ --fixed_htx_timestep ] arg   Use Fixed Time-Step for Haptics (default: 
                                    False)
  -a [ --load_multibody_files ] arg Description Filenames of Multi-Body(ies) to
                                    Launch, .e.g. -a <path>/test.yaml, 
                                    <another_path>/test2.yaml will load 
                                    multibodies test.yaml and test2.yaml if 
                                    they are valid files
  -l [ --load_multibodies ] arg     Index of Multi-Body(ies) to Launch, .e.g. 
                                    -l 1,2,3 will load multibodies at indexes 
                                    1,2,3. See launch.yaml file
  --launch_file arg                 Launch file path to load (default: 
                                    ../../ambf_models/descriptions/launch.yaml
  -g [ --show_gui ] arg             Show GUI
  --ns arg                          Override the default (or specified in ADF) 
                                    world namespace

------------------------------------------------------------

failed initialization
```

nag92, 2020-03-31:
Do you have a roscore running?

pacarvalho, 2020-03-31:
I do :( 

[image]

Any specific configuration I should give it? 

```
root@a88d8c19cf92:/usr/src/ambf/build# &roscore
bash: syntax error near unexpected token `&'
root@a88d8c19cf92:/usr/src/ambf/build# roscore &
[1] 142
root@a88d8c19cf92:/usr/src/ambf/build# ... logging to /root/.ros/log/7ee1e0cc-7351-11ea-b9c6-0242ac110003/roslaunch-a88d8c19cf92-142.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://a88d8c19cf92:46305/
ros_comm version 1.14.4

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.4

NODES

auto-starting new master
process[master]: started with pid [152]
ROS_MASTER_URI=http://a88d8c19cf92:11311/

setting /run_id to 7ee1e0cc-7351-11ea-b9c6-0242ac110003
process[rosout-1]: started with pid [163]
started core service [/rosout]
```

DhruvKoolRajamani, 2020-03-31:
You need to run the simulator in headless mode as the Docker file does not support an x interface (I'm assuming).

Append the argument with:
```bash
$ ./ambf_simulator -g off
```

pacarvalho, 2020-03-31:
@DhruvKoolRajamani That was it! :) 

Should probably add a descriptive error message for that situation. For, although obvious in retrospect, it is not a requirement that a new user might know. 

```
root@5ccc49d64b24:/usr/src/ambf/bin/lin-x86_64# ./ambf_simulator -g off

____________________________________________________________

ASYNCHRONOUS MULTI-BODY FRAMEWORK SIMULATOR (AMBF Simulator)

                (www.aimlab.wpi.edu)
                  (Copyright 2019)
____________________________________________________________

STARTUP COMMAND LINE OPTIONS: 

InputDevices Application Usage:
  -h [ --help ]                     Show help
  -n [ --ndevs ] arg                Number of Haptic Devices to Load
  -i [ --load_devices ] arg         Index number of devices to load which is 
                                    specified in input_device.yaml
  -e [ --enableforces ] arg         Enable Force Feedback on Haptic Devices
  -p [ --phx_frequency ] arg        Physics Update Frequency (default: 1000 Hz)
  -d [ --htx_frequency ] arg        Haptics Update Frequency (default: 1000 Hz)
  -t [ --fixed_phx_timestep ] arg   Use Fixed Time-Step for Physics (default: 
                                    False)
  -f [ --fixed_htx_timestep ] arg   Use Fixed Time-Step for Haptics (default: 
                                    False)
  -a [ --load_multibody_files ] arg Description Filenames of Multi-Body(ies) to
                                    Launch, .e.g. -a <path>/test.yaml, 
                                    <another_path>/test2.yaml will load 
                                    multibodies test.yaml and test2.yaml if 
                                    they are valid files
  -l [ --load_multibodies ] arg     Index of Multi-Body(ies) to Launch, .e.g. 
                                    -l 1,2,3 will load multibodies at indexes 
                                    1,2,3. See launch.yaml file
  --launch_file arg                 Launch file path to load (default: 
                                    ../../ambf_models/descriptions/launch.yaml
  -g [ --show_gui ] arg             Show GUI
  --ns arg                          Override the default (or specified in ADF) 
                                    world namespace

------------------------------------------------------------

Thread Joined: World
INFO! SETTING SIMULATION MAX ITERATIONS TO : 50
Thread Joined: light_left
Thread Joined: light_right
Thread Joined: Chassis
Thread Joined: ShockBL
Thread Joined: WheelBR
Thread Joined: ShockBR
Thread Joined: WheelBL
Thread Joined: ShockFL
Thread Joined: WheelFR
Thread Joined: ShockFR
Thread Joined: WheelFL
Found Lib: libdvrk_arm.so
No of dVRK Masters detected: 0

Running Headless (-g option provided) t = 1.2962e-05 sec
```

DhruvKoolRajamani, 2020-03-31:
Great! I'm sure they would be able to traceback to this issue if they come across the error.

pacarvalho, 2020-03-31:
@DhruvKoolRajamani True :)

commit 2eb5a75d (2020-03-31): Verbose output in case of failure to load graphics. Fixed #62
closed 2020-03-31 by adnanmunawar