WPI-AIM/ambf #67: How to use Python Client

alifahsanul, 2020-04-06:
Based on the wiki here https://github.com/WPI-AIM/ambf/wiki/The-Python-Client
We can use python to control the robot

Where should the python code be saved?
I tried putting it in my own folder (outside ambf folder) and it gives this error:

> 
Traceback (most recent call last):
  File "test_ambf.py", line 2, in <module>
    from ambf_client import Client
ImportError: No module named ambf_client
> 

I look at this issue https://github.com/WPI-AIM/ambf/issues/9
I don't understand this part
"source the ambf/build/devel folder to export ros_modules and then you should be able to import ambf_client"
Can you please elaborate?

Thank you

adnanmunawar, 2020-04-06:
You have to source the AMBF package path. As it says in the README, 

“ On Linux systems, please source the correct folder to achieve system wide availability of AMBF ROS modules. While in the build folder, you can run:

source ./devel/setup.bash

You can also permanently add the install location in your .bashrc with the following command:

echo "source ~/ambf/build/devel/setup.bash" >> ~/.bashrc”

Then it doesn’t matter where your python code is.

alifahsanul, 2020-04-06:
Thank you for the answer
Now i can import ambf_client in python

But now I got new error
Unable to register with master node [http://localhost:11311]: master may not be running yet. Will keep trying.

I have set
export ROS_MASTER_URI=http://localhost:11311

adnanmunawar, 2020-04-06:
Did you run roscore in a separate terminal? The warning is indicating that you have not.

alifahsanul, 2020-04-07:
Thank, it works after running roscore

closed 2020-04-07 by alifahsanul