WPI-AIM/ambf #69: Fix camera parenting specified via the ADF file

adnanmunawar, 2020-04-13:
Newer updates to the API have disabled camera parenting specified via the ADF file.

adnanmunawar, 2020-04-13:
@nsujumnong Please check this commit as it fixes the camera parenting issue we discussed last week. To help you understand better, open this file:

https://github.com/WPI-AIM/ambf/blob/master/ambf_models/descriptions/world/world.yaml#L48
and look at the data-block 
```yaml
# This example is to show how to parent w.r.t another body.
# When a parent is defined, the location, look at and up
# vector are with respect to the parent and not the world origin
cameraEndoscope:
  namespace: cameras/
  name: endo_camera
  location: {x: 0.0, y: 0.0, z: -0.1}
  look at: {x: 0.0, y: 0.0, z: -0.5}
  up: {x: 1.0, y: 0.0, z: 0.0}
  clipping plane: {near: 0.01, far: 10.0}
  field view angle: 0.8
  monitor: 0
  # stereo: {mode: Passive, eye separation: 0.22, focal length: 2.0}
  parent: /ambf/env/ecm/BODY remote center link
  # multipass: True

```

If you launch the ECM (with CLI -l6) now, you should see this camera parented to the ECM remote center link. Just make sure to add this camera in line 3:

https://github.com/WPI-AIM/ambf/blob/master/ambf_models/descriptions/world/world.yaml#L3

such that it becomes
```yaml
cameras: [cameras, cameraEndoscope]
```

nsujumnong, 2020-04-13:
Thanks, @adnanmunawar . I'll get to it right away.

nsujumnong, 2020-04-13:
@adnanmunawar  I just pulled and ran the updated code. However, the endo camera only spawn at the parent's frame but it seems that the camera is fixed in space and isn't actually attached to the frame. Does this happen on our end too?

adnanmunawar, 2020-04-13:
Interesting. And did you build the code? And did you run the code with the ECM?

nsujumnong, 2020-04-13:
My bad. I totally forgot about building the code first. I just did and now it's running perfectly. Sorry for not checking that first. Thanks for reminding me about it.

adnanmunawar, 2020-04-13:
No problem. Glad it's working now. Stay tuned or more updates. I am closing this issue now.

closed 2020-04-13 by adnanmunawar

cshreyastech, 2020-04-14:
@adnanmunawar , Thanks for the fix.

I added the to external/chai3d/CMakeLists.txt to successfully build. This seems to be due to one of the recent changes.
include_directories (${PROJECT_SOURCE_DIR}/extras)