WPI-AIM/ambf #72: set_joint_pos to 0 on model blender-kuka.yaml breaks model

Beccy97, 2020-05-13:
I'm using the python client to control the kuka robot example provided.
I'm executing the following code:

```python
# Import the Client from ambf_client package
from ambf_client import Client
import time

# Create a instance of the client
_client = Client()

# Connect the client which in turn creates callable objects from ROS topics
# and initiates a shared pool of threads for bi-directional communication
_client.connect()

print('\n\n----')
raw_input("We can see what objects the client has found. Press Enter to continue...")
# You can print the names of objects found. We should see all the links found
print(_client.get_obj_names())

# Lets get handles to the bodies of miro
arm_base_handle = _client.get_obj_handle('base')

print('\n\n----')
raw_input("Set all joint positions to 0, Press Enter to continue...")

for i in range(0, 7):
    arm_base_handle.set_joint_pos(i,0)
    time.sleep(3)

print('\n\n----')
raw_input("Let's clean up. Press Enter to continue...")
# Lastly to cleanup
_client.clean_up()
```
Up to joint 6, setting the joint positions to 0 works fine but for joint 7 the model breaks and parts are moving in an unintended way.

adnanmunawar, 2020-05-14:
Thanks for creating the issue. It is usually the case that the PID joint controller values are too high for this to happen. I can either update them or you can simply lower the gain values for the last joint where this is happening.If you are talking about the last joint, I believe it is this one

``` yaml
JOINT link6-link7:
  name: link6-link7
  parent: BODY link6
  child: BODY link7
  parent axis: {x: 0.0, y: -1.0, z: 0.0}
  parent pivot: {x: -0.003, y: -0.05, z: 0.053}
  child axis: {x: 0.0, y: 0.0, z: 1.0}
  child pivot: {x: 0.0, y: 0.0, z: 0.0}
  joint limits: {high: 3.054, low: -3.054}
  controller: {D: 4.0, I: 0, P: 2000.0}
  type: revolute
```

I would try lowering the D gain to 1.0 or lower and see if that affects your results. You might also wanna lower the P gain.

adnanmunawar, 2020-05-25:
Did the above answer solve your issue?

Beccy97, 2020-05-26:
As you suggested I lowered the D gain value for joint link6-link7 to 0.1 and the P gain to 100 in my local clone of the repository and now setting its joint position works as expected.
Thank you very much for your help

adnanmunawar, 2020-05-27:
You are welcome. We should probably add this to the documentation as this question is often asked. Closing this issue for now.

closed 2020-05-27 by adnanmunawar