WPI-AIM/ambf #73: 20.04 support

nag92, 2020-06-05:
When attempting to launch the simulator in 20.04 I get the following error:
```
Thread Joined: World
INFO! SETTING SIMULATION MAX ITERATIONS TO : 50
Thread Joined: light_left
Thread Joined: light_right
Thread Joined: default_camera
Thread Joined: Chassis
Thread Joined: ShockBL
Thread Joined: WheelBR
Thread Joined: ShockBR
Thread Joined: WheelBL
Thread Joined: ShockFL
Thread Joined: WheelFR
Thread Joined: ShockFR
Thread Joined: WheelFL
free(): double free detected in tcache 2
Aborted (core dumped)

```

nag92, 2020-06-08:
I think that it a memory leak problem. 

- https://stackoverflow.com/questions/57616404/what-does-it-mean-double-free-detected-in-tcache-2-while-using-mpz
- https://stackoverflow.com/questions/16698688/free-no-double-free-detected

I ran ambf_simulator with 
```bash
valgrind --leak-check=yes --log-file="leak.txt"   ./ambf_simulator 
```
Here is my log file from running it with those tags
[leak.txt](https://github.com/WPI-AIM/ambf/files/4748386/leak.txt)

This somewhat makes sense because the debug mode sets things to 0 where release mode had garbage collection problem or something like that. Why this problem is happening in 20.04 is still strange.

montematico, 2020-07-20:
Are any of the branches stable on Ubuntu 20.04 yet?

nag92, 2020-07-21:
> Are any of the branches stable on Ubuntu 20.04 yet?

no, I just tied on my VM running PopOs 20.04 on it still throws the same mem error

adnanmunawar, 2020-07-22:
I don't have much bandwidth these days to debug on Ubuntu 20.04, although based on a quick session with Nathaniel, we found that in Debug mode, there weren't any memory leak issues. This makes it super tricky to debug and thus I am postponing it for the moment.

adnanmunawar, 2020-09-21:
It appears that two methods with a `return` type of `bool` weren't returning anything if the intended tasks were successful. This still should cause the crash but apparently it did. Worthwhile for someone else to check as well if this works.

nag92, 2020-09-22:
interesting, ill check it soon

montematico, 2020-09-22:
> It appears that two methods with a `return` type of `bool` weren't returning anything if the intended tasks were successful. This still should cause the crash but apparently it did. Worthwhile for someone else to check as well if this works.

I just downloaded & built the ambf-1.0 branch on Kubuntu 20.04; It seems to be have built successfully and it is launching the demo objects without issue.

adnanmunawar, 2020-09-22:
Thanks for checking @montematico. I found out that not `returning` in `method/function` that expects a `return` is undefined behaviour. So we were just lucky previously where the crashes weren't happening. 

https://eel.is/c++draft/stmt.return

C++ Section 8.7.4

> Flowing off the end of a constructor, a destructor, or a non-coroutine function with a cv void return type is equivalent to a return with no operand. Otherwise, flowing off the end of a function other than main or a coroutine ([dcl.fct.def.coroutine]) results in undefined behavior.