WPI-AIM/ambf #75: Cannot get Geomagic Touch USB working with the AMBF simulator

DigeshC, 2020-06-15:
Hello,

We are trying to connect the Geomagic Touch USB with the ambf simulator but have been unsuccessful. We are currently working on a fresh install of Ubuntu 18.04. We think we have correctly set up the Openhaptics and Touch Drivers needed and correctly set up chai3d. This is the terminal message output.

`Thread Joined: World
INFO! SETTING SIMULATION MAX ITERATIONS TO : 50
Thread Joined: light_left
Thread Joined: light_right
Thread Joined: default_camera
Thread Joined: Chassis
Thread Joined: ShockBL
Thread Joined: WheelBR
Thread Joined: ShockBR
Thread Joined: WheelBL
Thread Joined: ShockFL
Thread Joined: WheelFR
Thread Joined: ShockFR
Thread Joined: WheelFL
****************************************
This is the DEVELOPER EDITION of OPENHAPTICS, commercial distribution is prohibited
Please contact 3D Systems to obtain a commercial license
****************************************
Found Lib: libdvrk_arm.so
No of dVRK Masters detected: 0
WARNING: FAILED TO LOAD DEVICE: "Geomagic Touch"`

Please advise us on how to go about this situation.

Thank you.

adnanmunawar, 2020-06-16:
Thanks for creating the issue. There are several ways around this. First let us confirm if your Geomagic is running properly. Can you clone this repo, build it, and then launch it. 

https://github.com/WPI-AIM/ros_geomagic/tree/master/geomagic_control

Can you confirm if the RViz GUI is updated when you move your Geomagic device? You can also simply echo the ROS topics that are launched when you run the above package and see if your joint positions are changing when the device is moved?

DigeshC, 2020-06-16:
Hello,

We were unsuccessful in building the package. When trying to install the package, we were met with this
```
lab@user:~/catkin_ws$ rosdep install --from-paths src --ignore-src -r -y
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
geomagic_control: Cannot locate rosdep definition for [openhaptics-ae]
Continuing to install resolvable dependencies...
#All required rosdeps installed successfully
```

And when we catkin_make the workspace, we were met with this error
```
lab@user:~/catkin_ws$ catkin_make
Base path: /home/lab/catkin_ws
Source space: /home/lab/catkin_ws/src
Build space: /home/lab/catkin_ws/build
Devel space: /home/lab/catkin_ws/devel
Install space: /home/lab/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/lab/catkin_ws/build"
####
####
#### Running command: "make -j8 -l8" in "/home/lab/catkin_ws/build"
####
[  0%] Built target geometry_msgs_generate_messages_cpp
[  0%] Built target geometry_msgs_generate_messages_nodejs
[  0%] Built target geometry_msgs_generate_messages_lisp
[  0%] Built target geometry_msgs_generate_messages_py
[  0%] Built target geometry_msgs_generate_messages_eus
[  0%] Built target _geomagic_control_generate_messages_check_deps_DeviceButtonEvent
[  0%] Built target _geomagic_control_generate_messages_check_deps_DeviceFeedback
[ 21%] Built target geomagic_control_generate_messages_py
[ 35%] Built target geomagic_control_generate_messages_nodejs
[ 50%] Built target geomagic_control_generate_messages_cpp
[ 64%] Built target geomagic_control_generate_messages_lisp
[ 85%] Built target geomagic_control_generate_messages_eus
[ 85%] Built target geomagic_control_generate_messages
[ 92%] Linking CXX executable /home/lab/catkin_ws/devel/lib/geomagic_control/device_node
/usr/bin/ld: cannot find -lncurses
collect2: error: ld returned 1 exit status
ros_geomagic/geomagic_control/CMakeFiles/device_node.dir/build.make:132: recipe for target '/home/lab/catkin_ws/devel/lib/geomagic_control/device_node' failed
make[2]: * [/home/lab/catkin_ws/devel/lib/geomagic_control/device_node] Error 1
CMakeFiles/Makefile2:1334: recipe for target 'ros_geomagic/geomagic_control/CMakeFiles/device_node.dir/all' failed
make[1]: * [ros_geomagic/geomagic_control/CMakeFiles/device_node.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: * [all] Error 2
Invoking "make -j8 -l8" failed
```
Please advise. Thank you.

DigeshC, 2020-06-16:
Hello
We were successfully able to build and launch the package.
We can confirm that the RViz GUI is updated when we move the Geomagic device.
Please let us know how we can continue. Thank you

adnanmunawar, 2020-06-16:
Perfect, so now since you have that working, you can spawn the device via an alternative interface. Here is how it works.

Please download this package.
https://github.com/adnanmunawar/geomagic_utils

It is just a Python script so no need to compile it. You can download it in your ROS workspace if you wish. 
1. After that, run your geomagic ros_node. You can simply run the headless node which does launch the RViz GUI.
```
roslaunch geomagic_control geomagic_headless.launch
```
2. With that running in the background. Run the Python file in the package based on the instructions in the README which are simply.
```
python mimic_mtms.py /<device_name>/ <publishing_frequency>
```
Where the <device_name> is the prefix that you may have given in step one. Refer to README https://github.com/adnanmunawar/geomagic_utils/blob/master/README.md for more details.

3. Now when you run your AMBF simulator, you should see the simulated gripper(s) represented via the Geomagic device. 

Using this method we are basically skipping the loading of Geomagic drivers in AMBF and instead tricking it to think that we have a dVRK MTM connected. 

Let me know if it doesn't work.

DigeshC, 2020-06-16:
So we got the script working and the simulator detects the haptic device as an MTMR <>.

However, when we try to use the input flag, -i, we have mixed responses,
-i 0 works and we can move the gripper around
-i 1 detects a device but we are unable to move the gripper

Moving forward, we just have to customize the MTM files instead of the Geomagic files while having the script running?

Thank you for all of your help

adnanmunawar, 2020-06-17:
Right, that is the expected behavior. The -i flag selects an input device in the order of devices listed in the `input devices`. This `input devices` file is set in the launch file as seen here: 

https://github.com/WPI-AIM/ambf/blob/master/ambf_models/descriptions/launch.yaml

```yaml
# This is the base file for Coordination Application
world config: ./world/world.yaml
# world config: ./world/raven_world.yaml
color config: ./color/colors.yaml
input devices config: ./input_devices/input_devices.yaml
multibody configs:
```
We can take a look at this ``input device`` file and its contents are as follows:
https://github.com/WPI-AIM/ambf/blob/master/ambf_models/descriptions/input_devices/input_devices.yaml

```yaml
input devices: [
  MTMR,
  MTML,
  Geomagic Touch,
  Razer Hydra0,
  Razer Hydra1,
  Novint Falcon1,
  Novint Falcon2]

MTML:
    hardware name: MTML
    haptic gain: {linear: 0.03, angular: 0.0}
    workspace scaling: 5
...
```
The `-i` flag selects the devices in the order of devices listed in the `input devices: []` data-block. So selecting `-i 0` means that AMBF will be looking for `MTMR` if it is present. And `-i 1` will be looking for 'MTML' if present. You can even add multiple devices as such `-i 0, 1, 2, 3`. If no -i flag is present, AMBF will look for all devices in the `input devices' data-block.

Now the point of confusion here is that the Python script which I pointed to earlier:
https://github.com/adnanmunawar/geomagic_utils/blob/master/mimic_mtms.py

This script can read two Geomagic devices and mimic them as `MTMR` and `MTML` proxy devices. Even though you may only have one Geomagic device, the script will still spawn the two MTM devices. Thus when you run AMBF, it will naturally think that two dVRK MTMs (Right and Left) are present. However, there is no Geomagic device to control the MTML proxy device and hence the result that you are seeing.

DigeshC, 2020-06-17:
Thank you for your help. It is now working as intended.

closed 2020-06-17 by DigeshC