WPI-AIM/ambf #76: use both vel and velocity(ies)

nag92, 2020-06-27:
In rigid body class half the functions use "vel" and the other half use "velocity". Should this be consistent? It would make it easier to have the same names style through out.

adnanmunawar, 2020-06-29:
Thanks for creating this issue. That is a good catch. I was creating some documentation the other day and noticed the inconsistency as well. Would fix it before merging back into master.

nag92, 2020-06-30:
which do you want? I can quickly resolve all the inconsistency.

adnanmunawar, 2020-06-30:
Thanks for taking this up. I suppose that since we are using “pos” instead of “position”, we should “vel” instead of “velocity”. There are some code examples in ambf_ros_modules that might need to be updated as well.

adnanmunawar, 2020-07-01:
Fixed in the feat-communication branch at 8b018738dc30e304e8bb02b3ae941d97da4e53da

PR #78 (2020-07-01): Fixing vel naming
fixes  #76
closed 2020-07-01 by adnanmunawar