WPI-AIM/ambf #77: Build fails in Ubuntu 18.04 and ROS melodic

JieYingWu, 2020-06-29:
Cmake finishes without any errors, though with warnings. 

The error trace from make is below. It seems unhappy about ROS. 
```

[ 91%] Built target actionlib_generate_messages_nodejs
Scanning dependencies of target ambf_world
[ 92%] Building CXX object ambf_ros_modules/ambf_comm/CMakeFiles/ambf_world.dir/src/World.cpp.o
In file included from /opt/ros/melodic/include/ros/node_handle.h:32:0,
                 from /opt/ros/melodic/include/ros/ros.h:45,
                 from /home/jieying/Research/ambf/ambf_ros_modules/ambf_comm/include/ambf_comm/RosComBase.h:48,
                 from /home/jieying/Research/ambf/ambf_ros_modules/ambf_comm/include/ambf_comm/WorldRosCom.h:48,
                 from /home/jieying/Research/ambf/ambf_ros_modules/ambf_comm/include/ambf_comm/World.h:49,
                 from /home/jieying/Research/ambf/ambf_ros_modules/ambf_comm/src/World.cpp:45:
/opt/ros/melodic/include/ros/publisher.h:53:52: warning: defaulted and deleted functions only available with -std=c++11 or -std=gnu++11
     Publisher& operator=(const Publisher& other) = default;
                                                    ^
In file included from /opt/ros/melodic/include/ros/node_handle.h:33:0,
                 from /opt/ros/melodic/include/ros/ros.h:45,
                 from /home/jieying/Research/ambf/ambf_ros_modules/ambf_comm/include/ambf_comm/RosComBase.h:48,
                 from /home/jieying/Research/ambf/ambf_ros_modules/ambf_comm/include/ambf_comm/WorldRosCom.h:48,
                 from /home/jieying/Research/ambf/ambf_ros_modules/ambf_comm/include/ambf_comm/World.h:49,
                 from /home/jieying/Research/ambf/ambf_ros_modules/ambf_comm/src/World.cpp:45:
/opt/ros/melodic/include/ros/subscriber.h:52:52: warning: defaulted and deleted functions only available with -std=c++11 or -std=gnu++11
   Subscriber& operator=(const Subscriber& other) = default;
                                                    ^
In file included from /opt/ros/melodic/include/ros/node_handle.h:34:0,
                 from /opt/ros/melodic/include/ros/ros.h:45,
                 from /home/jieying/Research/ambf/ambf_ros_modules/ambf_comm/include/ambf_comm/RosComBase.h:48,
                 from /home/jieying/Research/ambf/ambf_ros_modules/ambf_comm/include/ambf_comm/WorldRosCom.h:48,
                 from /home/jieying/Research/ambf/ambf_ros_modules/ambf_comm/include/ambf_comm/World.h:49,
                 from /home/jieying/Research/ambf/ambf_ros_modules/ambf_comm/src/World.cpp:45:
/opt/ros/melodic/include/ros/service_server.h:51:58: warning: defaulted and deleted functions only available with -std=c++11 or -std=gnu++11
   ServiceServer& operator=(const ServiceServer& other) = default;
                                                          ^
In file included from /opt/ros/melodic/include/ros/node_handle.h:35:0,
                 from /opt/ros/melodic/include/ros/ros.h:45,
                 from /home/jieying/Research/ambf/ambf_ros_modules/ambf_comm/include/ambf_comm/RosComBase.h:48,
                 from /home/jieying/Research/ambf/ambf_ros_modules/ambf_comm/include/ambf_comm/WorldRosCom.h:48,
                 from /home/jieying/Research/ambf/ambf_ros_modules/ambf_comm/include/ambf_comm/World.h:49,
                 from /home/jieying/Research/ambf/ambf_ros_modules/ambf_comm/src/World.cpp:45:
/opt/ros/melodic/include/ros/service_client.h:49:58: warning: defaulted and deleted functions only available with -std=c++11 or -std=gnu++11
   ServiceClient& operator=(const ServiceClient& other) = default;
                                                          ^
In file included from /opt/ros/melodic/include/ros/node_handle.h:36:0,
                 from /opt/ros/melodic/include/ros/ros.h:45,
                 from /home/jieying/Research/ambf/ambf_ros_modules/ambf_comm/include/ambf_comm/RosComBase.h:48,
                 from /home/jieying/Research/ambf/ambf_ros_modules/ambf_comm/include/ambf_comm/WorldRosCom.h:48,
                 from /home/jieying/Research/ambf/ambf_ros_modules/ambf_comm/include/ambf_comm/World.h:49,
                 from /home/jieying/Research/ambf/ambf_ros_modules/ambf_comm/src/World.cpp:45:
/opt/ros/melodic/include/ros/timer.h:52:42: warning: defaulted and deleted functions only available with -std=c++11 or -std=gnu++11
   Timer& operator=(const Timer& other) = default;
                                          ^
In file included from /opt/ros/melodic/include/ros/node_handle.h:38:0,
                 from /opt/ros/melodic/include/ros/ros.h:45,
                 from /home/jieying/Research/ambf/ambf_ros_modules/ambf_comm/include/ambf_comm/RosComBase.h:48,
                 from /home/jieying/Research/ambf/ambf_ros_modules/ambf_comm/include/ambf_comm/WorldRosCom.h:48,
                 from /home/jieying/Research/ambf/ambf_ros_modules/ambf_comm/include/ambf_comm/World.h:49,
                 from /home/jieying/Research/ambf/ambf_ros_modules/ambf_comm/src/World.cpp:45:
/opt/ros/melodic/include/ros/wall_timer.h:52:50: warning: defaulted and deleted functions only available with -std=c++11 or -std=gnu++11
   WallTimer& operator=(const WallTimer& other) = default;
                                                  ^
In file included from /opt/ros/melodic/include/ros/node_handle.h:39:0,
                 from /opt/ros/melodic/include/ros/ros.h:45,
                 from /home/jieying/Research/ambf/ambf_ros_modules/ambf_comm/include/ambf_comm/RosComBase.h:48,
                 from /home/jieying/Research/ambf/ambf_ros_modules/ambf_comm/include/ambf_comm/WorldRosCom.h:48,
                 from /home/jieying/Research/ambf/ambf_ros_modules/ambf_comm/include/ambf_comm/World.h:49,
                 from /home/jieying/Research/ambf/ambf_ros_modules/ambf_comm/src/World.cpp:45:
/opt/ros/melodic/include/ros/steady_timer.h:52:54: warning: defaulted and deleted functions only available with -std=c++11 or -std=gnu++11
   SteadyTimer& operator=(const SteadyTimer& other) = default;
                                                      ^
In file included from /home/jieying/Research/ambf/ambf_ros_modules/ambf_comm/include/ambf_comm/RosComBase.h:50:0,
                 from /home/jieying/Research/ambf/ambf_ros_modules/ambf_comm/include/ambf_comm/WorldRosCom.h:48,
                 from /home/jieying/Research/ambf/ambf_ros_modules/ambf_comm/include/ambf_comm/World.h:49,
                 from /home/jieying/Research/ambf/ambf_ros_modules/ambf_comm/src/World.cpp:45:
/opt/ros/melodic/include/tf/tf.h:354:8: error: ‘shared_ptr’ in namespace ‘std’ does not name a template type
   std::shared_ptr<tf2_ros::Buffer> getTF2BufferPtr() { return tf2_buffer_ptr_;};
        ^
/opt/ros/melodic/include/tf/tf.h:403:8: error: ‘shared_ptr’ in namespace ‘std’ does not name a template type
   std::shared_ptr<tf2_ros::Buffer> tf2_buffer_ptr_;
        ^
/opt/ros/melodic/include/tf/tf.h: In member function ‘ros::Duration tf::Transformer::getCacheLength()’:
/opt/ros/melodic/include/tf/tf.h:331:43: error: ‘tf2_buffer_ptr_’ was not declared in this scope
   ros::Duration getCacheLength() { return tf2_buffer_ptr_->getCacheLength();}
                                           ^
/opt/ros/melodic/include/tf/tf.h: In member function ‘void tf::Transformer::setUsingDedicatedThread(bool)’:
/opt/ros/melodic/include/tf/tf.h:349:46: error: ‘tf2_buffer_ptr_’ was not declared in this scope
   void setUsingDedicatedThread(bool value) { tf2_buffer_ptr_->setUsingDedicatedThread(value);};
                                              ^
/opt/ros/melodic/include/tf/tf.h: In member function ‘bool tf::Transformer::isUsingDedicatedThread()’:
/opt/ros/melodic/include/tf/tf.h:351:42: error: ‘tf2_buffer_ptr_’ was not declared in this scope
   bool isUsingDedicatedThread() { return tf2_buffer_ptr_->isUsingDedicatedThread();};
                                          ^
ambf_ros_modules/ambf_comm/CMakeFiles/ambf_world.dir/build.make:62: recipe for target 'ambf_ros_modules/ambf_comm/CMakeFiles/ambf_world.dir/src/World.cpp.o' failed
make[2]: *** [ambf_ros_modules/ambf_comm/CMakeFiles/ambf_world.dir/src/World.cpp.o] Error 1
CMakeFiles/Makefile2:1822: recipe for target 'ambf_ros_modules/ambf_comm/CMakeFiles/ambf_world.dir/all' failed
make[1]: *** [ambf_ros_modules/ambf_comm/CMakeFiles/ambf_world.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2

```

adnanmunawar, 2020-06-29:
Thanks for creating the issue Jie. I am guessing this is from the feat-communication branch? Can you also paste the output of the  `cmake` command so that I can verify you compiler etc..

JieYingWu, 2020-06-29:
I get the error in both feat_communication and master it seems. The trace from cmake is:

```
-- Boost version: 1.65.1
-- Using CATKIN_DEVEL_PREFIX: /home/jieying/Research/ambf/build/devel
-- Using CMAKE_PREFIX_PATH: /home/jieying/catkin_ws/devel;/opt/ros/melodic
-- This workspace overlays: /home/jieying/catkin_ws/devel;/opt/ros/melodic
-- Found PythonInterp: /home/jieying/anaconda3/envs/sofa/bin/python2 (found suitable version "2.7.16", minimum required is "2") 
-- Using PYTHON_EXECUTABLE: /home/jieying/anaconda3/envs/sofa/bin/python2
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/jieying/Research/ambf/build/test_results
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Found gmock sources under '/usr/src/googletest': gmock will be built
CMake Warning (dev) at /usr/src/googletest/CMakeLists.txt:3 (project):
  Policy CMP0048 is not set: project() command manages VERSION variables.
  Run "cmake --help-policy CMP0048" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  The following variable(s) would be set to empty:

    PROJECT_VERSION
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at /usr/src/googletest/googlemock/CMakeLists.txt:40 (project):
  Policy CMP0048 is not set: project() command manages VERSION variables.
  Run "cmake --help-policy CMP0048" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  The following variable(s) would be set to empty:

    PROJECT_VERSION
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at /usr/src/googletest/googletest/CMakeLists.txt:47 (project):
  Policy CMP0048 is not set: project() command manages VERSION variables.
  Run "cmake --help-policy CMP0048" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  The following variable(s) would be set to empty:

    PROJECT_VERSION
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found PythonInterp: /home/jieying/anaconda3/envs/sofa/bin/python2 (found version "2.7.16") 
-- Using Python nosetests: /home/jieying/anaconda3/envs/sofa/bin/nosetests-2.7
-- catkin 0.7.26
-- BUILD_SHARED_LIBS is on
-- *** FOUND ROS ON THIS MACHINE, ENABLING SUPPORT FOR AMBF_ROS MODULES
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- ambf_msgs: 16 messages, 0 services
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
CMake Warning (dev) at external/tf_function/CMakeLists.txt:2 (project):
  Policy CMP0048 is not set: project() command manages VERSION variables.
  Run "cmake --help-policy CMP0048" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  The following variable(s) would be set to empty:

    PROJECT_VERSION
This warning is for project developers.  Use -Wno-dev to suppress it.

-- *** FOUND AMBF_COMM, AMBF_MSGS and DVRK_ARM, ENABLING SUPPORT
-- *** FOUND OPENCV
-- *** FOUND ROS_CV_BRIDGE and IMAGE_TRANSPORT, ENABLING SUPPORT
RAZER HYDRA'S ROS PACKAGE NOT FOUND
USINGX86_64_ASM
-- Boost version: 1.65.1
-- Found the following Boost libraries:
--   program_options
--   filesystem
--   system
-- Boost version: 1.65.1
-- Found the following Boost libraries:
--   program_options
--   filesystem
--   system
-- Configuring done
CMake Warning at ambf_simulator/CMakeLists.txt:47 (add_executable):
  Cannot generate a safe runtime search path for target ambf_simulator
  because files in some directories may conflict with libraries in implicit
  directories:

    runtime library [libpython2.7.so.1.0] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/jieying/anaconda3/envs/sofa/lib

  Some of these libraries may not be found correctly.

CMake Warning at ambf_controller/CMakeLists.txt:48 (add_executable):
  Cannot generate a safe runtime search path for target ambf_controller
  because files in some directories may conflict with libraries in implicit
  directories:

    runtime library [libpython2.7.so.1.0] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/jieying/anaconda3/envs/sofa/lib

  Some of these libraries may not be found correctly.

-- Generating done
-- Build files have been written to: /home/jieying/Research/ambf/build

```

adnanmunawar, 2020-06-29:
Alright, I see two possible issues here, an obvious one is that somehow C++11 is not being used, hence the issue with std::shared_ptr. The other may be emanating from anaconda? Let's deal with this issue via Teamviewer and I shall update this issue based on what we find.

adnanmunawar, 2020-06-29:
It appears that the system had an older compiler GCC 5.x which wasn't compiling the `ambf_ros_modules` package using the `C++11` compile flags. The immediate fix was to force this by uncommenting this line:
https://github.com/WPI-AIM/ambf/blob/master/ambf_ros_modules/ambf_comm/CMakeLists.txt#L47
```
add_compile_options(-std=c++11)
```
We should permanently uncomment this line as we are using C++11 features in the code now.

closed 2020-06-29 by adnanmunawar