WPI-AIM/ambf #79: ambf_client issue: base_obj referenced before assignment

qiguaiderenmen, 2020-07-01:
When I try to run client.connect() function an error occurr:

Traceback (most recent call last):
  File "/home/kamerider/ambf_walker/PendTesting/first_test.py", line 14, in <module>
    _client.connect()
  File "<string>", line 196, in connect
  File "<string>", line 123, in create_objs_from_rostopics
UnboundLocalError: local variable 'base_obj' referenced before assignment

This variable is referenced in line 123, while it is actually first assigned value in line 127. How can we deal with this issue?

adnanmunawar, 2020-07-01:
This file `/home/kamerider/ambf_walker/PendTesting/first_test.py` doesn't seem to be part of the AMBF repo. Is this your own code? You can paste it here so I can see what's going on.

nag92, 2020-07-01:
I am running the same code and it works on fine on my machine. 
```python
#!/usr/bin/env python
import sys
from os import sys, path

from Model import DoublePendulm
import rospy
from ambf_client import Client

_client = Client()
_client.connect() # its failing hereon Simon machine, im not sure why. 
rate = rospy.Rate(1000)
pend = DoublePendulm.DoublePendulum(_client)
while 1:
    pass

```

adnanmunawar, 2020-07-01:
Do you have a GitHub link for this code which I can use to reproduce the issue as the information is not sufficient for me to look at anything.

adnanmunawar, 2020-07-01:
Oh I think I got it. See if the latest commit fixes the issue.

qiguaiderenmen, 2020-07-01:
So much apologies for my late response! Yeah the actual error line is in ambf/ambf_ros_modules/ambf_comm/scripts/ambf_client.py, sorry for my ambiguous description. Also the new commit has already fixed this issue, thanks a lot for your help!!

commit c2e6f7cb (2020-07-01): Should fix #79
closed 2020-07-01 by qiguaiderenmen