WPI-AIM/ambf #89: Camera endoscope

elisaiovene, 2020-10-02:
Hello,

I attached the camera on the endoscope (BODY remote center link). 
In the world.yaml file I have:

[image]

plus I included 'cameraEndoscope' on the cameras field (third row). 

However when I run the simulation, the scene moves all around:

[image]

Is there a way to get a more stable and fixed view?

adnanmunawar, 2020-10-03:
Thanks for creating the question. Can you explain a bit more what you mean by "the scene moves all around" and requiring a "fixed view". Are you connecting the endoscopic link to a haptic device?

elisaiovene, 2020-10-05:
Maybe in the future. For now I just wrote a Python client code to move it to a determined position.  Once it reaches that position, I would like the endoscope to be fixed but it keeps moving and so does the camera attached to it. 
Thanks!

adnanmunawar, 2020-10-06:
Hmm, can you share the Python script so I may have a look at it? The endoscope shouldn't move if it is commanded via joint positions.

elisaiovene, 2020-10-06:
Here it is: 
```ruby
from ambf_client import Client
import time

c = Client()
c.connect()

ecm = c.get_obj_handle('baselink')
ecm.set_joint_vel(4, -0.1) #tool goes back
time.sleep(1.0)
ecm.set_joint_vel(1, 1.0) #pitch link goes up
time.sleep(1.0)
ecm.set_joint_vel(4, 0.1) #tool gets inside the abdomen 
time.sleep(1.0)

input('clean up, press enter to continue..')
c.clean_up()
```
Setting the joint position was worst, that's why I used the velocity instead.

adnanmunawar, 2020-10-06:
Thanks for pasting the code as it clarifies things a bit. 
Two points:
1) I think you may be using the ECM model from the older code (ambf master branch). The ecm model in the ambf-1.0 branch is updated and more stable. It also has only the four joints that the actual ECM does. The remaining joints are passive and do not communicate so the control via Python client is much cleaner. I would recommend using that model instead.

2) With the updated model the positon control should be stable. However, if you still prefer to use velocity, then keep reading the current joint positions via `get_joint_pos` after setting the velocity and then when you think that the robot has reached your intended joint position, use the recorded joint positions to override the velocity command by issuing `set_joint_pos`. 

Also, if the `cleanup()` is called, the watchdog safety timers will expire thus removing the position and velocity controllers and setting the effort on each joint to 0. As a result, the robot would fall under the weight of gravity.

elisaiovene, 2020-10-07:
Perfect! It works definitely better! 
Thank you

closed 2020-10-08 by adnanmunawar