WPI-AIM/ambf #90: Haptic feedback

vanessaloia, 2020-10-02:
Good afternoon, 
__
I connected the haptic device to the robot in simulation. 
To get the haptic feedback, I enabled it with the -e 1 flag and, in _the ambf/ambf_framework/afInputDevice.h_ file, I changed the _m_dev_force_enable_ from false to true. 
However when the robot interacts with other bodies in the scene I don't get any feedback. Is there something else I need to change to get it? 

Moreover, is there a way to plot the forces of interaction when a collision happens?

adnanmunawar, 2020-10-03:
Thanks for creating the issue. Which exact haptic device are you talking about? You do not have to manually enable the `m_dev_force_enable` flag as it shall be set automatically. 
What you might need to change is the gains, here is an example:

``` yaml
MTMR:
  hardware name: MTMR
  haptic gain: {linear: 0.03, angular: 1}
  workspace scaling: 5
  simulated multibody: "../multi-bodies/grippers/pr2 gripper small red.yaml"
  # root link:
  location: {
    position: {x: -0.5, y: 0, z: 0},
    orientation: {r: 0, p: 0, y: 0}}
  button mapping: {
    a1: 1,
    a2: 2,
    next mode: 3,
    prev mode: 4}
  pair cameras: [camera1, camera2] # The motion of the IID-SDE is w.r.t the first camera in the list
```
For the haptic device, the field `haptic gain` is relevant. Try turning up the `linear gain`, but set a low gain as currently there isn't much safety while commanding forces.

Also, what branch are you on? If on `master` I would advise switching to `ambf-1.0`.

vanessaloia, 2020-10-05:
Yes, I'm on `ambf-1.0` 
Anyway, I changed the haptic gain and now it is working 
Is there a way to plot the interaction force? 
Thanks!

adnanmunawar, 2020-10-06:
Yes, we did have support for that, however, I commented out that code to enable integration with ambf-1.0. I can try re-integrating that with the branch

vanessaloia, 2020-10-06:
It would be great! Thanks