WPI-AIM/ambf #96: Publishing a geometry_msg in the /Command topic

elisaiovene, 2020-10-16:
Hello, 

I'm writing a position-position controller to move the Kuka robot that I have in the AMBF simulator through the haptic device Sigma7. 
I have created a ros node ` /sigma` from which I subscribe the amount of motion as a geometry_msg::Pose and I want to apply it to the kuka robot by publishing this msg in the topic `/ambf/env/kuka_tip/Command` (where kuka_tip is the end-effector). 

When I run the script I get this error:
`[ERROR] [1602851968.149073971]: Client [/ambf_comm_node] wants topic /ambf/env/kuka_tip/Command to have datatype/md5sum [ambf_msgs/RigidBodyCmd/c54fc48da5e84240a367eff7725e74b4], but our version has [geometry_msgs/Pose/e45d45a5a1ce597b249e23fb30fc871f]. Dropping connection.`

Do you have any suggestion? 
Thanks

adnanmunawar, 2020-10-16:
Thanks for creating the issue. ambf expects a **RigidBodyCmd** message type to command a rigid body. I suppose you are giving it a **Pose** message type. You will need to use the correct message type and I think this is what the ERROR message is complaining about. You may want to take a look at this example and it's readme on how we use the **RidigBodyCommand** message to control joint-space and cartesian-space.

https://github.com/WPI-AIM/ambf/tree/ambf-1.0/ambf_ros_modules/examples/object_control_example

elisaiovene, 2020-10-16:
Thanks for your answer! 
I changed the type of the msg I was publishing. 
Now I have a publisher `Kuka_tip_pose = n.advertise < ambf_msgs::RigidBodyCmd > ("/ambf/env/kuka_tip/Command",1000);`that publishes an ambf_msgs::RigidBodyCmd and the error has disappeared.
However, when I run the simulation, the end-effector does not move. 
Do you recommend another type of msg or another topic to publish to, to control the end effector? 

Thanks

adnanmunawar, 2020-10-16:
Did you set the appropriate controller type in the message? If you look at the message below, 

``` yaml
Header header
int8 TYPE_FORCE=0
int8 TYPE_POSITION=1
int8 TYPE_VELOCITY=2
int8 cartesian_cmd_type # The type of cartesian cmd i.e. F/P/V
geometry_msgs/Pose pose
geometry_msgs/Wrench wrench
geometry_msgs/Twist twist
int8[] joint_cmds_types # For each joint, the type of cmd i.e. F/P/V
float32[] joint_cmds
bool publish_children_names
bool publish_joint_names
bool publish_joint_positions
```

The field `int8 cartesian_cmd_type` needs to be set appropriately. 0 for Force, 1 for Pose and 2 for velocity control. If you have your own PID controller, you can use Force or Velocity. If you want ambf_framework to internally compute the PID, then use the Pose.

elisaiovene, 2020-10-16:
It worked, thanks!

elisaiovene, 2020-10-22:
Hello! I'm replying to this issue because it still regards `ambf_msgs`. 
I need to record the joint position so I have a subscriber that subscribes to the /State topic. 
In the `RigidBodyState.msg` there is a `float32[] joint positions`, how can I get to that data? 
I tried but I got a segmentation fault but maybe because I'm accessing the data in the wrong way. 

Thanks!

adnanmunawar, 2020-10-22:
Is your code in C++ or Python? If Python, please refer to this folder.
https://github.com/WPI-AIM/ambf/tree/ambf-1.0/ambf_ros_modules/examples/object_control_example
This GUI lets you control an Object in Cartesian Mode (Position, Force, Velocity) as well as its joints in Joint Space Mode (Position, Effort, Velocity). This code utilizes the `ambf_client` which automatically sets up the appropriate flags for a body to publish its children's names, joint names, joint position,s and velocities. If using the Client, you can get the joint positions by using the appropriate methods.

If instead, you are writing your code in C++, a body's communication handler does not publish the names, positions and velocities of its joints unless asked to. There are a couple of different ways you can ask for it. In your case, you can set the following field in its `/Command` message and then wait for a little while (< 1.0 second) before the body's communication handler starts to publish these things.

``` yaml
Header header
int8 TYPE_FORCE=0
int8 TYPE_POSITION=1
int8 TYPE_VELOCITY=2
int8 cartesian_cmd_type # The type of cartesian cmd i.e. F/P/V
geometry_msgs/Pose pose
geometry_msgs/Wrench wrench
geometry_msgs/Twist twist
int8[] joint_cmds_types # For each joint, the type of cmd i.e. F/P/V
float32[] joint_cmds
bool publish_children_names
bool publish_joint_names
bool publish_joint_positions
```
Refer to the last three flags i.e. 

```
bool publish_children_names
bool publish_joint_names
bool publish_joint_positions
```
Setting them to true will result in the appropriate data being published.

elisaiovene, 2020-10-22:
Yes, I set the `bool joint positons` to true in the ` /Command` topic I'm publishing to. 
However when I try to subscribe to the topic /State of the same link to have the joint position data, I get a segmentation fault. 
What I do is: 
```ruby
ambf_msgs::RigidBodyState pos_kuka;
void kuka_poseCallback(const ambf_msgs::RigidBodyState::ConstPtr& msg)
{
pos_kuka.joint_positions = msg->joint_positions;
}
```
Any suggestion?

adnanmunawar, 2020-10-22:
Do you get a segfault in AMBF or your own code?

elisaiovene, 2020-10-22:
My own code

adnanmunawar, 2020-10-22:
There may be many reasons for why that would occur but I won't be able to help you but looking at just the 3 lines of code you posted above.

elisaiovene, 2020-10-23:
This is the part of the main relative to AMBF:
```ruby
#include <stdio.h>
#include <math.h>
#include <pthread.h>
#include <sstream>
#include "My_Sigma_class.hpp"
#include "ambf_msgs/RigidBodyCmd.h"
#include "ambf_msgs/RigidBodyState.h"

using namespace std;
ambf_msgs::RigidBodyState pos_kuka;

void kuka_poseCallback(const ambf_msgs::RigidBodyState::ConstPtr& msg){
        pos_kuka.pose.position.x = msg->pose.position.x;
	pos_kuka.pose.position.y = msg->pose.position.y;
	pos_kuka.pose.position.z = msg->pose.position.z;
	pos_kuka.pose.orientation.x = msg->pose.orientation.x;
	pos_kuka.pose.orientation.y = msg->pose.orientation.y;
	pos_kuka.pose.orientation.z = msg->pose.orientation.z;
        pos_kuka.joint_positions = msg->joint_positions;
        std::cout<<"joint: "<<pos_kuka<<std::endl;
}
int main(int argc, char** argv) {
	std::string nodename("sigma");
	if (argc > 1)
	   nodename = argv[1];
	ros::init(argc, argv, nodename);
	ros::NodeHandle n(nodename);
	//Defining publishers
	ros::Publisher Kuka_tip_pose;
	//Defining subscriber
	ros::Subscriber kuka_pose;

	Kuka_tip_pose = n.advertise < ambf_msgs::RigidBodyCmd > ("/ambf/env/kuka_tip/Command",1000);
        kuka_pose  = n.subscribe("/ambf/env/kuka_tip/State",1000, kuka_poseCallback);

        geometry_msgs::Pose pos0;
	geometry_msgs::Pose  posf; 
	ambf_msgs::RigidBodyCmd delta;
        while (ros::ok()) {
       	   if(sigma[i].getPedal().data == 1){
           	dhdGetPositionAndOrientationFrame(&(temp_state.p[0]), &(temp_state.p[1]), &(temp_state.p[2]), temp_state.orientationMatrix,i);
                dhdGetOrientationRad(&last_orientation[0], &last_orientation[1],&last_orientation[2]);
                posf.position.x =  temp_state.p[0];
	        posf.position.y =  temp_state.p[1];
		posf.position.z =  temp_state.p[2];
		posf.orientation.x =  last_orientation[0];
		posf.orientation.y =  last_orientation[1];
		posf.orientation.z =  last_orientation[2];
                
	       delta.pose.position.x = abs(pos0.position.x - posf.position.x) + pos_kuka.pose.position.x;
               delta.pose.position.y = abs(pos0.position.y - posf.position.y) + pos_kuka.pose.position.y;
               delta.pose.position.z = abs(pos0.position.z - posf.position.z) + pos_kuka.pose.position.z;
               delta.pose.orientation.x = abs(pos0.orientation.x - posf.orientation.x) + pos_kuka.pose.orientation.x;
               delta.pose.orientation.y = abs(pos0.orientation.y - posf.orientation.y) + pos_kuka.pose.orientation.y;
               delta.pose.orientation.z = abs(pos0.orientation.z - posf.orientation.z) + pos_kuka.pose.orientation.z;           
               delta.cartesian_cmd_type = 1; 
               delta.publish_joint_positions = true;              
               Kuka_tip_pose.publish(delta);
            }
            else{
               	drdStart();
	        drdStop(true);
	    }

	ros::spinOnce();
	loop_rate.sleep();
    }
}
```

adnanmunawar, 2020-10-23:
You are using the body named 'kuka_tip' to get the joint positions of the robot. This would not work. To get all the joint positions, you will need a handle to 'base' or whatever the **root parent** body of the robot is.

adnanmunawar, 2020-10-23:
I can't run this code as is because I don't have the file "My_sigma_class.hpp" nor do I have the sigma device. However, commenting out those irrelevant parts, I do not get a segfault error as you described. Try executing the code through a debugger and get an idea of exactly where the code fails.

elisaiovene, 2020-10-26:
I get the segmentation fault when I try to access to the single joint position. 
For example `pos_kuka.joint_positions[0]`. 
However, I'll try to follow your advice. 
Thank you!

adnanmunawar, 2020-10-26:
Oh so this line of code was not in the code that you shared and this in fact would give your SEGFAULT. The reason being that `kuka_tip` being the distal end-effector does not have any joints to report. In AMBF, only the `parents` have information about the joints and joint names. So for example, if you wish to get the joint positions, you would need to subscribe / publish to the 'base' link.

elisaiovene, 2020-10-26:
I modified that already, now I have the sub and pub referring to the baselink: 
```ruby
ros::Subscriber joint_pose_sub;
ros::Publisher joint_pose_pub;
joint_pose_sub = n.subscribe("/ambf/env/base/State",1000, joint_poseCallback);
joint_pose_pub   = n.advertise < ambf_msgs::RigidBodyCmd > ("/ambf/env/base/Command",1000);

void joint_poseCallback(const ambf_msgs::RigidBodyState::ConstPtr& msg){
    pos_joint.joint_positions = msg->joint_positions;
    pos_joint.joint_velocities = msg->joint_velocities;
    pos_joint.joint_efforts = msg->joint_efforts;
    std::cout<<"joint: "<<pos_joint<<std::endl;
}
```
and in the main I have set the bool to true for publishing the joint position. 
From the std::cout I don't get any data, moreover when I try to store the single positions of the joints, I get the segfault.

adnanmunawar, 2020-10-26:
Please paste the full relevant code that I can use to reproduce the issue rather than trying to explain it.

elisaiovene, 2020-10-27:
```ruby 

#include <stdio.h>
#include <math.h>
#include <pthread.h>
#include <sstream>
#include "My_Sigma_class.hpp"
#include "ambf_msgs/RigidBodyCmd.h"
#include "ambf_msgs/RigidBodyState.h"

using namespace std;
ambf_msgs::RigidBodyState pos_joint;
ambf_msgs::RigidBodyState pos_kuka;
void kuka_poseCallback(const ambf_msgs::RigidBodyState::ConstPtr& msg){
    pos_kuka.pose.position.x = msg->pose.position.x;
    pos_kuka.pose.position.y = msg->pose.position.y;
    pos_kuka.pose.position.z = msg->pose.position.z;
    pos_kuka.pose.orientation.x = msg->pose.orientation.x;
    pos_kuka.pose.orientation.y = msg->pose.orientation.y;
    pos_kuka.pose.orientation.z = msg->pose.orientation.z;
}
void joint_poseCallback(const ambf_msgs::RigidBodyState::ConstPtr& msg){
    pos_joint.joint_positions = msg->joint_positions;
    pos_joint.joint_positions[1] = msg->joint_position[1];
    std::cout<<"joint: "<<pos_joint<<std::endl;
}
int main(int argc, char** argv) {
    std::string nodename("sigma");
    if (argc > 1)
	nodename = argv[1];
    ros::init(argc, argv, nodename);
    ros::NodeHandle n(nodename);

    ros::Publisher joint_pose_pub;
    ros::Publisher Kuka_tip_pose;
    ros::Subscriber joint_pose;
    ros::Subscriber kuka_pose;
    joint_pose  = n.subscribe("/ambf/env/base/State",1000, joint_poseCallback);
    joint_pose_pub  = n.advertise < ambf_msgs::RigidBodyCmd > ("/ambf/env/base/Command",1000);
    Kuka_tip_pose = n.advertise < ambf_msgs::RigidBodyCmd > ("/ambf/env/kuka_tip/Command",1000);
    kuka_pose = n.subscribe("/ambf/env/kuka_tip/State",1000, kuka_poseCallback);
    ambf_msgs::RigidBodyCmd joint_pub;
    geometry_msgs::Pose pos0;
    geometry_msgs::Pose  posf; 
    ambf_msgs::RigidBodyCmd delta; 
    while (ros::ok()) {
         dhdGetPositionAndOrientationFrame(&(temp_state.p[0]), &(temp_state.p[1]), &(temp_state.p[2]), temp_state.orientationMatrix,i);
         dhdGetOrientationRad(&last_orientation[0], &last_orientation[1],&last_orientation[2]);
               
           posf.position.x =  temp_state.p[0];
	   posf.position.y =  temp_state.p[1];
	   posf.position.z =  temp_state.p[2];
	   posf.orientation.x =  last_orientation[0];
	   posf.orientation.y =  last_orientation[1];
	   posf.orientation.z =  last_orientation[2];
	   delta.pose.position.x = -(pos0.position.x - posf.position.x) + pos_kuka.pose.position.x;
           delta.pose.position.y = -(pos0.position.y - posf.position.y) + pos_kuka.pose.position.y;
           delta.pose.position.z = -(pos0.position.z - posf.position.z) + pos_kuka.pose.position.z;
           delta.pose.orientation.x = (pos0.orientation.x - posf.orientation.x) + pos_kuka.pose.orientation.x;
           delta.pose.orientation.y = (pos0.orientation.y - posf.orientation.y) + pos_kuka.pose.orientation.y;
           delta.pose.orientation.z = (pos0.orientation.z - posf.orientation.z) + pos_kuka.pose.orientation.z;
           delta.cartesian_cmd_type = 1; 
           Kuka_tip_pose.publish(delta);
           joint_pub.publish_joint_positions = true;
           ros::spinOnce();
	   loop_rate.sleep();

	}
}
```

I also tried to run your code, what happened is that the joint positions get published but the model of the Kuka disappears.

adnanmunawar, 2020-10-27:
The logic in the pasted code is incorrect. You shall have to change the joint_poseCallback method to something of the sort below

```cpp
    pos_joint = *msg;
    for (int i = 0 ; i < msg->joint_positions.size() ; i++){
        std::cerr << "jp: [" << i << "]: " << msg->joint_positions.at(i) << std::endl;
    }
```

OR, you would need to put a sleep call for a second or so before the **ros::spinOnce()** call in your main function. There are additional flaws in your code that have to do with how you are setting the orientation. The orientation here is a Quaternion, NOT an Euler rotation.

Regarding your comment "I also tried to run your code, what happened is that the joint positions get published but the model of the Kuka disappears.". Did you run the example code snipped AS IT IS, or did you make modifications to it?

adnanmunawar, 2020-11-24:
Closing this issue as it seems to have gone sideways from the topic. Please feel free to open a new issue.

closed 2020-11-24 by adnanmunawar