WPI-AIM/ambf #99: Error: what(): vector::_M_range_check: __n (which is 0) >= this ->size() (whic is 0)

Debora-b, 2020-11-04:
Hi, I would like to publish the new joint configuration of the robot but  I get the error mentioned above.

this is part of the code: 

`ambf_msgs::RigidBodyCmd pos_joint_new;`
`for(int j=0; j<pos_joint.joint_positions.size(); j++){  `                                               
          `joint_positions_new[j] = delta_q[j] + pos_joint.joint_positions.at(j);`
`pos_joint_new.joint_cmds_types.push_back(1);`
`pos_joint_new.joint_cmds.at(j) = joint_positions_new[j];`
`}`
`joint_pub.publish(pos_joint_new);`

adnanmunawar, 2020-11-05:
Please provide more information about what robot you're running and which body's handle are you subscribed to etc.

adnanmunawar, 2022-02-17:
Closing stale issue.

closed 2022-02-17 by adnanmunawar