WPI-AIM/ambf #102: dVRK model

jiaqixuac, 2020-11-21:
Hi,

It's an amazing work for ambf.
I wonder if the dVRK model in this project consistent with https://github.com/WPI-AIM/dvrk_env.
Also, is the dVRK model consistent with the https://github.com/jhu-dvrk/dvrk-ros/tree/master/dvrk_model and the real world one.
Where can we find the inertial parameters?

Looking forward to your reply.

adnanmunawar, 2020-11-23:
Hello Jiaqixuac,
Thank you!

The kinematic parameters and frame assignment (joint directions) are identical to the dVRK Manual and thus https://github.com/jhu-dvrk/dvrk-ros/tree/master/dvrk_model for MTM, PSM, and ECM. As far as the inertial parameters are concerned, we kind of left it on the back burner to update them for the AMBF models and thus there are not similar to https://github.com/WPI-AIM/dvrk_env. 

Just so you know, the parameters computed here https://github.com/WPI-AIM/dvrk_dynamics_identification are accurate and I do not remember if we ever put these parameters in the https://github.com/WPI-AIM/dvrk_env. Hopefully, we can update the parameters on the AMBF models soon enough and I shall post on this thread once we do so.

jiaqixuac, 2020-11-24:
Thanks a lot for your reply. So the parameters in AMBF are accurate. I will look at them in detail.