WPI-AIM/ambf #103: Interfacing Ambf with dvrk

SimoGa, 2020-12-04:
Hi, I actually interfaced the dvrk with ambf , I realized that the movements are not corresponding along x and y (if I go forward with the MTMR the gripper goes backwards), so I would like to rotate the reference system of the gripper with rot  (z, 180 °).  This thing should be changed in CdvrkDevices.cpp right?
Beside i've seen that i can add a second camera for stereovision, where do i insert the device topic?
Thanks

adnanmunawar, 2020-12-08:
I shall continue the discussion from our email exchange here. You may want to try to newest commit to check if the linear and angular mapping has been fixed. 

Regarding your second question, can you elaborate more on what the issue or question is?

SimoGa, 2020-12-08:
Yes thank you for your reply. I explain my second question better because i've understood how to rotate the reference frame , as you mentioned in the email. From my inspection, i understood that more than a camera can be exploited during the simulation, how this can be done? I have only found in input_device.yaml the possibility to modify the pair cameras. 
Thank you again

adnanmunawar, 2020-12-08:
Hmm, thanks for explaining your question. So are you asking about how to add another camera to the simulation? If that is the question, then you can do the following. Head over the **world.yaml** file as shown here

https://github.com/WPI-AIM/ambf/blob/ambf-1.0/ambf_models/descriptions/world/world.yaml

On top of this file, you should see the following fields

```yaml
enclosure size: {length: 10.0, width: 10.0, height: 3.0}
lights: [light1]
cameras: [camera1]
environment: "./checkered_floor.yaml"
namespace: /ambf/env/
max iterations: 50
gravity: {x: 0.0, y: 0.0, z: -9.81}
```

You can add a camera to the list of cameras, for example:
```yaml
enclosure size: {length: 10.0, width: 10.0, height: 3.0}
lights: [light1]
cameras: [camera1, camera2]
environment: "./checkered_floor.yaml"
namespace: /ambf/env/
max iterations: 50
gravity: {x: 0.0, y: 0.0, z: -9.81}
```

The **camera2** in this example is already defined later in the same file as shown here

```yaml
camera2:
  namespace: cameras/
  name: openhmd
  location: {x: 3.0, y: 0.0, z: 2.0}
  look at: {x: 0.0, y: 0.0, z: -1.0}
  up: {x: 0.0, y: 0.0, z: 1.0}
  clipping plane: {near: 0.01, far: 10.0}
  field view angle: 0.8
  stereo: {mode: Passive, eye separation: 0.22, focal length: 2.0}
  monitor: 3
```

You can of course create your own camera with your desired parameters and add it to the list of **cameras** above.

closed 2020-12-08 by SimoGa