WPI-AIM/ambf #107: joints not appearing in ECM model with no passive joints

nag92, 2020-12-26:
in the ECM model I made all the joints 
passive: false

This made 9/12 joints appear in the output. The following joints did not appear

- yaw link-pitch back link
- pitch bottom link-pitch end link
- pitch top link-pitch end link

adnanmunawar, 2021-01-04:
@nag92 @cshreyastech Hey guys, can you create an answer for this based on our last discussion.

cshreyastech, 2021-01-04:
@adnanmunawar , thanks for explaining the behavior. 

             A
         /        \
        /           \
      B           C
        \          /
         \        /
             D
              |
              |
             E
Suppose that we have 2 parallel joints ABD and ACD. Only one of the joint would be published in AMBF say ABD in this example. This is because body D can be controlled by just ACD and commanding ABD will become redundant. 

The joint which is defined first in the YAML file among AB or AC gets the privileged of ROS control.

So, what if we would like to get the joint details of body C or Joint CD?
This can be done by creating a rigidbody handler of body C. Joint CD can be controlled by getting the child of C.