WPI-AIM/ambf #112: Cannot get Falcon USB working with the AMBF simulator

haotiangu, 2021-01-24:
Hi Adnan,

We are trying to connect the Falcon USB with the ambf simulator but have been unsuccessful. We have loaded the falcon and PSM in simulator, but we could not to have actual falcon to control the movement of PSM in simulator. We guess it is because short of installation about the driver of falcon and node which is responsible for communicating between actual falcon and PSM in slmulator. We followed up tutorial of [ros_falcon](https://github.com/WPI-AIM/ros_falcon) to set up ros_falcon and falcon driver. The communication between actual did not work. So please tell me what should we do next.  

[image]

adnanmunawar, 2021-01-25:
Hello Haotian,
Thanks for creating the issue. Can you elaborate more on the issue. From the figure above, it seems that you do have the Novint Falcon detected by the simulator, and as a result, a free-floating gripper spawned in the simulation. Is that right?

haotiangu, 2021-01-25:
Hi Adnan,

Thanks for your timely response. Correct. Based on the input_device.yaml, I need to hang 5 after -i to load the falcon.
`input devices: [
  MTMR,
  MTML,
  Geomagic Touch,
  Razer Hydra0,
  Razer Hydra1,
  Novint Falcon1,
  Novint Falcon2]
`

I just followed up your tutorial and run ./ambf_simulator -i 5 -l 4. The number 5 hang after -i means that falcon is input_device. As you can see from picture, the falcon device have been loaded in simulator. But it seems like when I tried to move actual falcon to control the movement of PSM in simulator. It did not work.
[image]

[image]

 Could you please give me some tips to let falcon work with AMBF simulator.  Thank you so much,

Best,
Haotian

adnanmunawar, 2021-01-25:
Thanks for the clarification and I am glad that you are able to figure out the command-line options. Is it correct that when you move the Falcon device, the free-floating gripper (the gripper with the green base and black fingers) moves? 
It should and that indicates that the Falcon is paired to the free-floating gripper. Please read this entry that I recently added to the Wiki:

https://github.com/WPI-AIM/ambf/wiki/Input-Devices

To summarize, for teleoperating the dVRK PSM via an Input Device, you would want to control the robot in joint space rather than directly commanding the end-effector in Cartesian space as is the case with the free-floating gripper. To do so, it would be easier to use the Python client.

Here is an example where I am using the ROS interface for a Phantom Omni to control the PSM while NOT loading the device into the simulator directly (setting the option `-n 0` for the ambf_simulator, you can read more about CLI options here https://github.com/WPI-AIM/ambf/wiki/Command-Line-Arguments).

Here is the example repo now:
https://github.com/adnanmunawar/surgical_robotics_challenge/blob/master/scripts/master_slave_control.py

Basically, you run the PSM in this repo as follows:
1. `./ambf_simulator --launch_file <surgical_robotics_challenge>/launch.yaml -l 12`.
2. Separately, run the `roslaunch` file for the Phantom Omni in a different terminal. This, in our case, is from this repo (https://github.com/WPI-AIM/ros_geomagic). This creates ROS topics for the device which we can use to read its Pose and command forces if required.
3. Then we run the script shared above. This script then takes the input device's pose and computes the IK for the PSM and then sets the computed joint positions via the Python client.

In your case, you will have to run the Falcon ROS node and create a python class similar to (https://github.com/adnanmunawar/surgical_robotics_challenge/blob/master/scripts/geomagic_device.py) to wrap the functionality. You will also need to edit the (https://github.com/adnanmunawar/surgical_robotics_challenge/blob/master/scripts/master_slave_control.py) file to include the Falcon class instead. 

Hope this helps.

haotiangu, 2021-01-26:
Hi Adnan,

Thanks for your elaborate explanation. Those information is very helpful for me to understand some sub-function of ambf package.
Actually, when I move the Falcon device, the free-floating gripper did not move. I am figuring out how to let free-floating gripper to move by using falcon control.

About the falcon ros node, I have found the [ros_falcon](https://github.com/WPI-AIM/ros_falcon) your github and compiled it successfully. But it seems like there is no launch available in this node. Please give me some tips about how to use ros_falcon in my case. Thank you so much.

Best,
Haotian

adnanmunawar, 2021-01-26:
I see. It appears that the Falcon device may not be initialized and/or homed. I updated the wiki of that package to help you with the process. If you have any further questions regarding that package, please feel free to open an issue there.

haotiangu, 2021-01-26:
Hi Adnan,

Thank you so much for your timely response. Thanks for updating the tutorial to set up falcon driver. I followed up tutorial on your github. I got stuck by the problem which is shown in picture when I ran `rosrun ros_falcon driver` Looking forward to hearing your guidance. Thank you so much.

[image]

Best,
Haotian

haotiangu, 2021-01-26:
Hi Adnan,

Sorry for bothering you again. To be more specific about what I got stuck, It seems like that I could not sence the locking force when pulling the knob all the way outwards and then pushing it inwards over and over again. That might be reason why I can not initialize or home the falcon device.

adnanmunawar, 2021-01-26:
No worries, can you try running the `findfalcons` program from the `libnifalcon` package and see what the terminal output is?

haotiangu, 2021-01-26:
[image]

haotiangu, 2021-01-26:
As you can see from picture, that is the output when I ran findfalcons.

adnanmunawar, 2021-01-27:
Great. And can you also try the program called ‘barrow_mechanics’ and confirm if you can sense the haptic sphere within the workspace?

haotiangu, 2021-01-27:
Hi Adnan,

Thanks for your timely response. After I ran barrow_mechancis, the output from terminal looks like picture displayed below. Please allow me ask you a question for clarification. What does that mean that sense the haptic sphere within the workspace.

[image]
Looking forward to your response.

Best,
Haotian

haotiangu, 2021-01-27:
When I pulling the knob all the way outwards and then pushing it inwards again and again while run utility(findfalcon or barrow_mechanics) function, I could not sense a locking force. 
I was wondering is there any problem about my falcon device. Or is there any problem about my operation when I homed or initiated the falcon device.

adnanmunawar, 2021-01-27:
The virtual haptic sphere a potential field that is located at the center of the device's workspace. If you move the knob around that region you would feel as if there was a spherical-shaped collision object located there and you can only traverse along its surface but not push inside of it.

We can also test something else. While you run the program `findfalcons` and the terminal starts printing, do the values of Enc1, Enc2, and Enc3 change when you move the knob around?

haotiangu, 2021-01-27:
Correct. The value of Enc1,Enc2,and Enc3 change when I move the knob around.

adnanmunawar, 2021-01-27:
Hmm, I am not sure then why the ros_falcon package, nor spawning the device in ambf doesn't seem to work. It might easier to debug this on a call rather than a lengthy back and forth discussion here.

haotiangu, 2021-01-27:
Thanks for your patience. When I ran rosrun ros_falcon driver, the output from terminal looks like picture displayed below.
It seems like I also need to do some work to initialize or home the falcon.
[image]

adnanmunawar, 2021-01-27:
Great so it seems that the ros_falcon is able to detect the device. Can you perform a rostopic echo on the position topic spawned by the ros_falcon driver node and verify if the position message's data changes when you move the device. Also, you can try running ambf with the device (make sure to terminate the ros node beforehand) and see if you are able to move the simulated gripper after homing the device?

haotiangu, 2021-01-27:
It looks like the falcon pos did not publish. It may be that I did not initialize the falcon properly.
[image]

haotiangu, 2021-01-27:
This picture is about rostopic list.
[image]

adnanmunawar, 2021-01-27:
I think you may be echoing the incorrect topic. Can you echo the /falconPos rather than ros_falcon/falconPos.

haotiangu, 2021-01-27:
Oh. Sorry about my fault. The terminal output looks like the picture displayed below.
[image]

adnanmunawar, 2021-01-27:
You need to source the ROS workspace where the ros_falcon pkg was built.

haotiangu, 2021-01-27:
ah. Thank you so much. The position in terminal is changing when I move the falcon knob all around.
The result looks like in picture displayed below.
[image]

adnanmunawar, 2021-01-27:
That’s good to hear. Thank you for sharing all the terminal images. Can you now try with AMBF, close the ROS node first. And if the simulated gripper doesn’t move, perform the homing procedure.

haotiangu, 2021-01-27:
Hi Adnan,

Thank you so much for your help.
Follow up your guidance. I tried to run the ./ambf_simulator
The result from terminal looks like picture displayed below. When I moved the falcon knob, the gripper in simulator also move correspondingly.
[image]

adnanmunawar, 2021-01-27:
You’re welcome and I’m glad it is working as expected.

adnanmunawar, 2021-01-27:
The issue seems to be resolved thus closing it.

closed 2021-01-27 by adnanmunawar