WPI-AIM/ambf #114: opening chrome causes instability

nag92, 2021-02-12:
The model become unstable when I open chrome. If I have no chrome windows open the model is stable and will hold a point/follow a trajectory fine. When I open a chrome window the model starts to jerk around. This might be a hardware limitation of my laptop. I can can post a video if you want.

adnanmunawar, 2021-02-20:
Thanks for creating the issue. I think it would be good to post that video here along with some explanation. I would like to see if opening the Chrome process afters roscore, or the rospy, so testing this using the ambf_client_cpp would help make some things clear.

nag92, 2021-02-22:
Here is the video of the C++ client working. Keep in mind I am running a minimal example. The Python code works slightly differently: https://www.dropbox.com/s/1fpfu2i1d6ccjbf/IMG_0045.mp4?dl=0

Also here are some timing graphs for a "clean" 60s of control at 500Hz. Meaning the computer was doing nothing else. I just had the AMBF simulation, a terminal, and my IDE. 
C++: 
[image]

Python: 
[image]

As the graphs show the std is much tighter for the C++ controller. @cshreyastech  this is good validation for the C++ client.

cshreyastech, 2021-02-23:
@nag92 , nice finding. thanks a lot for this effort.

nag92, 2021-02-23:
its a very weird problem.

nag92, 2021-02-23:
[image]

nag92, 2021-02-23:
I am going to do some more exmanation of what is causing the timing lag

nag92, 2021-02-24:
the timing of sending a cmd is weirdly  multi-modal

adnanmunawar, 2021-03-21:
Hey guys thanks for the details on this issue. Other than our discussions on slack about this being related to higher values of D gain, I wonder if this is somehow related to Issue #121 . Can you you pull the latest version and check if the issues persists with everything else the same.

nag92, 2021-03-21:
of course, need some time to care of some other tasks