WPI-AIM/ambf #117: moving Kuka robot with sigma7

Zimmo95, 2021-02-25:
Hello,
I'm new on AMBF and I'm trying to move the KUKA phantom in the AMBF environment using the sigma7 robot.
I succeded to move the pr2-gripper-small-red using the sigma 7 after have modified the files "..ambf/ambf_models/descriptions/input_devices/input_devices.yaml" , "ambf/ambf_framework/afInputDevices.cpp" and "ambf/ambf_framework/afInputDevices.h".
Later, I modified the simulated multibody of sigma.7 in "ambf/ambf_models/descriptions/input_devices/input_devices.yaml" with "../multi-bodies/robots/blender-kuka.yaml" but kuka started to spin all his joints in a chaotic way without stopping and without allowing me to control him.

Thank you for your availability,
Best, 
Enrico

Zimmo95, 2021-03-01:
Hi Adnan, 
Thanks for your time.
I added the example and modified the ./input_devices/input_devices.yaml file writing in the first position
" sigma7:
  hardware name: sigma.7
[...]

 root link: "link7"  " since I don't have the tip

Then I typed:
#1 roscore
#2 cd <ambf>/ambf_ros_modules/examples/gripper_control_examples
python gripper_control_via_device_ifc.py  // it open correctly the sliders
#3 ./ambf_simulator -p60 -t1 --launch_file ../../ambf_ros_modules/examples/input_device_teleop_example/launch.yaml -l8 -i0
// it doesn't work and the terminal prints 
"
INFO! WORLD CONFIG PATH: ../../ambf_ros_modules/examples/input_device_teleop_example/./world 
Thread Joined: World
INFO! SETTING SIMULATION MAX ITERATIONS TO : 50
Thread Joined: light_left
Thread Joined: camera1
ERROR, REQUESTED MULTI-BODY IDX 8 HOWEVER, 0 INDEXED MULTI-BODIES DEFINED
Found Lib: libdvrk_arm.so
No of dVRK Masters detected: 1
CONNECTING TO: MTMR
Thread Joined: simulated_device_1/MTMR
WARNING: PHYSICAL DEVICE : "MTMR" LOCATION NOT DEFINED 
WARNING: CAN'T FIND ANY OBJECTS NAMED: "camera2" IN GLOBAL MAP 
Existing OBJECTS in Map: 1
camera1
WARNING: CAN'T FIND ANY OBJECTS NAMED: "camera2" IN GLOBAL MAP 
Existing OBJECTS in Map: 1
camera1
Segmentation fault (core dumped)

So, i tried to write 
#3 ./ambf_simulator -p60 -t1  -l8 -i0  // it works and opens the window but i cannot move the kuka with the sigma7 and it prints 
"INFO! WORLD CONFIG PATH: ../../ambf_models/descriptions/./world 
Thread Joined: World
INFO! SETTING SIMULATION MAX ITERATIONS TO : 50
Thread Joined: light_left
Thread Joined: default_camera
Thread Joined: base
Thread Joined: link1
Thread Joined: link2
Thread Joined: link3
Thread Joined: link4
Thread Joined: link5
Thread Joined: link6
Thread Joined: link7
Found Lib: libdvrk_arm.so
No of dVRK Masters detected: 1
CONNECTING TO: MTMR
[Exception]: bad file
ERROR! FAILED TO LOAD ADF FILE: ../../ambf_models/descriptions/./input_devices/../blender-kuka-tip-sphere.yaml
WARNING: FAILED TO LOAD DEVICE: "sigma7"
 

Thanks for your availability
Enrico

Zimmo95, 2021-03-02:
Hi Adnan, 
Thanks for your time.
Everything works now!

Best,
Enrico

adnanmunawar, 2021-03-02:
Glad it worked out for you. Would you mind posting what exactly you did that made it work so that people may refer to this issue if they want to get a Sigma7 up and running?

Zimmo95, 2021-03-04:
Sure, 
In ./input_devices/input_devices.yaml, I commented the line  "simulated multibody" and modified the line  "root link" writing  root link: "link7", since I don't have the tip.
Then, I changed the name of "hardware name" in sigma7 and positioned it in first position (-i0).
Finally, I ran cd <ambf>/bin/lin-x86_64
./ambf_simulator -p60 -t1 -l8 -i0

Enrico

closed 2021-03-04 by adnanmunawar