WPI-AIM/ambf #122: Drilling process

Matteo-0, 2021-03-16:
Hi Adnan,
I am trying to simulate in Ambf the process of drilling a vertebral bone and trying to get the force exchanged in the interface between the bone and the cutter, so that a robot, that is performing the drilling can stop when the force goes beyond a certain threshold.
I would like to know if this kind of simulation is possible to be implemented in Ambf.

Thanks for your availability,

Matteo

adnanmunawar, 2021-03-19:
Hi Matteo,
How do you intend to drill the bone and/or visualize the bone drilling? The other aspects that you describe are present in the current implementation.

Matteo-0, 2021-03-20:
Hi Adnan,
Thanks for your reply!

I would like to visualize something similar to what is shown in the first video that you can see in the following link:
https://www.reddit.com/r/Unity3D/comments/c8177p/was_making_this_drill_press_simulation_and/?utm_source=share&utm_medium=web2x&context=3

Matteo

adnanmunawar, 2021-03-21:
Hi Matteo,
Thanks for sharing the link to the video and it looks really cool. Unfortunately, we do not have this implemented in AMBF at the moment. The closest thing that we do have is an ongoing integration of voxel-based drilling, most of the implementation of which is imported from CHAI-3D (an external project used in AMBF). I don't think I can give an exact timeline of how long it is going to take but at least a few months.