WPI-AIM/ambf #124: cmake not working on MacOS

22agulati, 2021-04-02:
Hi Adnan,

Hope all's well.

I am working on a project with UC Berkeley's AUTOLab and have been struggling to complete the build process (cmake is not functioning on my Mac). Wanted to see if there's any fix to this. To be specific, I get the following error:

CMake Error at CMakeLists.txt:92 (find_package):
  By not providing "Findyaml-cpp.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "yaml-cpp",
  but CMake did not find one.

  Could not find a package configuration file provided by "yaml-cpp" with any
  of the following names:

    yaml-cppConfig.cmake
    yaml-cpp-config.cmake

  Add the installation prefix of "yaml-cpp" to CMAKE_PREFIX_PATH or set
  "yaml-cpp_DIR" to a directory containing one of the above files.  If
  "yaml-cpp" provides a separate development package or SDK, be sure it has
  been installed.

-- Configuring incomplete, errors occurred!
See also "/Users/agulati/desktop/Research/ambf/build/CMakeFiles/CMakeOutput.log".
See also "/Users/agulati/desktop/Research/ambf/build/CMakeFiles/CMakeError.log".

Thanks1

adnanmunawar, 2021-04-02:
Hello,
Thank you for the introduction. On Macintosh, you should be able to install the dependencies such as `Boost` and `YAML-CPP` using homebrew. Here is a link to yaml-cpp on homebrew:

https://formulae.brew.sh/formula/yaml-cpp

Just a heads-up though, the ROS modules are not built on Mac so you wouldn't be able to use the Python Client or any of the ROS interfaces. You could install ROS on your Mac to enable them, but we have not tested this out on our end.

adnanmunawar, 2021-05-07:
Is this issue still there?

closed 2022-04-29 by adnanmunawar