WPI-AIM/ambf #125: Build Problem

nnagururu, 2021-04-03:
Hello!

I'm getting the following error when I run make:

> In file included from /home/nnagururu/ambf/ambf_framework/afFramework.cpp:45:0:
> /home/nnagururu/ambf/ambf_framework/afFramework.h:1639:10: error: ‘mutex’ in namespace ‘std’ does not name a type
>      std::mutex m_mutex;
>           ^~~~~
> CMakeFiles/ambf_framework.dir/build.make:75: recipe for target 'CMakeFiles/ambf_framework.dir/ambf_framework/afFramework.cpp.o' failed
> make[2]: *** [CMakeFiles/ambf_framework.dir/ambf_framework/afFramework.cpp.o] Error 1
> CMakeFiles/Makefile2:117: recipe for target 'CMakeFiles/ambf_framework.dir/all' failed
> make[1]: *** [CMakeFiles/ambf_framework.dir/all] Error 2
> Makefile:90: recipe for target 'all' failed
> make: *** [all] Error 2

This is my output for cmake:

> CMake Warning (dev) at /usr/local/share/cmake-3.20/Modules/FindOpenGL.cmake:315 (message):
>   Policy CMP0072 is not set: FindOpenGL prefers GLVND by default when
>   available.  Run "cmake --help-policy CMP0072" for policy details.  Use the
>   cmake_policy command to set the policy and suppress this warning.
> 
>   FindOpenGL found both a legacy GL library:
> 
>     OPENGL_gl_LIBRARY: /usr/lib/x86_64-linux-gnu/libGL.so
> 
>   and GLVND libraries for OpenGL and GLX:
> 
>     OPENGL_opengl_LIBRARY: /usr/lib/x86_64-linux-gnu/libOpenGL.so
>     OPENGL_glx_LIBRARY: /usr/lib/x86_64-linux-gnu/libGLX.so
> 
>   OpenGL_GL_PREFERENCE has not been set to "GLVND" or "LEGACY", so for
>   compatibility with CMake 3.10 and below the legacy GL library will be used.
> Call Stack (most recent call first):
>   CMakeLists.txt:93 (find_package)
> This warning is for project developers.  Use -Wno-dev to suppress it.
> 
> -- Found Boost: /usr/include (found version "1.65.1")  
> CMake Warning at CMakeLists.txt:104 (find_package):
>   By not providing "Findcatkin.cmake" in CMAKE_MODULE_PATH this project has
>   asked CMake to find a package configuration file provided by "catkin", but
>   CMake did not find one.
> 
>   Could not find a package configuration file provided by "catkin" with any
>   of the following names:
> 
>     catkinConfig.cmake
>     catkin-config.cmake
> 
>   Add the installation prefix of "catkin" to CMAKE_PREFIX_PATH or set
>   "catkin_DIR" to a directory containing one of the above files.  If "catkin"
>   provides a separate development package or SDK, be sure it has been
>   installed.
> 
> 
> USINGX86_64_ASM
> -- Found Boost: /usr/include (found version "1.65.1") found components: program_options filesystem system 
> -- Configuring done
> -- Generating done
> -- Build files have been written to: /home/nnagururu/ambf/build

I checked and I'm pretty sure catkin is up to date.

> ros-melodic-catkin is already the newest version (0.7.29-1bionic.20201014.191449).

I'm running Ubuntu 18.04.5 LTS on Parallels

Thanks in advance!

adnanmunawar, 2021-04-03:
Hi Animesh,
Thanks for posting this issue. Can you try with the latest commit fd3bfc60ad7a0b1b7567d7a4e65fa49c49183883.

Regarding the second issue with catkin, do you have ROS sourced? I.e. in your .bashrc file or in your terminal?

nnagururu, 2021-04-04:
Hi, 

I think those suggestions solved the build.

However, when I try to launch the simulator using the guidance in wiki I get:

> INFO! WORLD CONFIG PATH: ../../ambf_models/descriptions/./world 
> [ERROR] [1617554176.955098937]: [registerPublisher] Failed to contact master at [localhost:11311].  Retrying...

adnanmunawar, 2021-04-04:
Yes, you should start roscore in a separate terminal so that the communication server can initialize. It’s also convenient to launch the roscore once and keep it running in the background rather than running it every time you start and close AMBF.

nnagururu, 2021-04-04:
Ahhh, thank you. I'm up and running

closed 2021-04-04 by nnagururu