WPI-AIM/ambf #128: Video in Rviz

nnagururu, 2021-04-10:
Unable to see video in Rviz, see error messages below:

[image]

The depth data renders, but not the camera.

adnanmunawar, 2021-04-11:
Thanks for creating the issue. Did you set (uncomment) the `publish image` flag for the camera in the world.yaml file?

nnagururu, 2021-04-11:
Yes publish image is set, see world.yaml file below:

```
enclosure size: {length: 10.0, width: 10.0, height: 3.0}
lights: [light1, light2]
cameras: [camera1]
environment: "./checkered_floor.yaml"
namespace: /ambf/env/
max iterations: 50
gravity: {x: 0.0, y: 0.0, z: -9.81}
# shaders:
#   path: ../../ambf_shaders/normal_map
#   vertex: "shader.vs"
#   fragment: "shader.fs"
# skybox: # Uncomment to load this skybox
#   path: ./skybox/ocean_and_mountains
#   right: px.png
#   left:  nx.png
#   top: py.png
#   bottom: ny.png
#   front: pz.png
#   back: nz.png
#   shaders:
#     path: ./skybox/
#     vertex: "shader.vs"
#     fragment: "shader.fs"

light1:
  namespace: /ambf/env/lights/
  name: light1
  location: {x: 0.0, y: 0.5, z: 2.5}
  direction: {x: 0, y: 0, z: -1.0}
  spot exponent: 0.3
  shadow quality: 0
  cutoff angle: 0.7

light2:
  namespace: lights/
  name: light2
  location: {x: 2, y: 2, z: 5}
  direction: {x: 0, y: 0, z: -1.0}
  spot exponent: 0.3
  shadow quality: 5
  cutoff angle: 1.2

light3:
  location: {x: 0.5, y: 0.0, z: -0.5}
  direction: {x: 0, y: 0, z: -1.0}
  spot exponent: 0.3
  shadow quality: 0
  cutoff angle: 0.7
  parent: /ambf/env/BODY LightCone

camera1:
  namespace: cameras/
  name: default_camera
  location: {x: 4.0, y: 0.0, z: 2.0}
  look at: {x: 0.0, y: 0.0, z: -0.5}
  up: {x: 0.0, y: 0.0, z: 1.0}
  clipping plane: {near: 0.1, far: 10.0}
  field view angle: 0.8
  monitor: 0
  # multipass: True
  publish image: True
  # publish image interval: 1 # Publish every nth scene update
  publish depth: True
  # publish depth noise: {mean: 0.1, std_dev: 0.01, bias: 0.0}
  # publish depth interval: 1 # Publish every nth scene update
  # publish image resolution: {width: 640, height: 480}
  depth shaders:
    path: ../../../ambf_shaders/depth
    vertex: "shader_old.vs"
    fragment: "shader_old.fs"

# This example is to show how to parent w.r.t another body.
# When a parent is defined, the location, look at and up
# vector are with respect to the parent and not the world origin
cameraEndoscope:
  namespace: cameras/
  name: endo_camera
  location: {x: 0.0, y: 0.0, z: -0.1}
  look at: {x: 0.0, y: 0.0, z: -0.5}
  up: {x: 1.0, y: 0.0, z: 0.0}
  clipping plane: {near: 0.01, far: 10.0}
  field view angle: 0.8
  monitor: 0
  # stereo: {mode: Passive, eye separation: 0.22, focal length: 2.0}
  parent: /ambf/env/ecm/BODY remote center link
  # multipass: True

camera2:
  namespace: cameras/
  name: openhmd
  location: {x: 3.0, y: 0.0, z: 2.0}
  look at: {x: 0.0, y: 0.0, z: -1.0}
  up: {x: 0.0, y: 0.0, z: 1.0}
  clipping plane: {near: 0.01, far: 10.0}
  field view angle: 0.8
  stereo: {mode: Passive, eye separation: 0.22, focal length: 2.0}
  monitor: 3
  # parent: /ambf/env/BODY Occulus
  # orthographic view width: 6
  # device: Razer Hydra

camera3:
  namespace: cameras/
  name: normal_camera
  location: {x: -3.0, y: 0.0, z: 2.0}
  look at: {x: 0.0, y: 0.0, z: -0.5}
  up: {x: 0.0, y: 0.0, z: 1.0}
  clipping plane: {near: 0.01, far: 10.0}
  field view angle: 0.8
  monitor: 2
  # parent: /ambf/env/BODY Occulus
  # orthographic view width: 6
# device: Razer Hydra
```

adnanmunawar, 2021-04-12:
Great, so I think the problem may have been the topic type that was chosen to be displayed in RViz. Instead of selecting the "Camera" type in */ImageData, you should select the "Image" . Here is a screenshot of what this selection would look like.

[image]

nnagururu, 2021-04-12:
That worked, I can see the image now. I've searched online but I can't seem to figure out how to overlay the image and the point cloud to get an image like you have. The following is what I currenlty see:

[image]

adnanmunawar, 2021-04-12:
The depth point cloud that AMBF streams already has the color data encoded into each point. To view this in Rviz, you can set the **Color Transformer** to RGB8 as shown here:

[image]

nnagururu, 2021-04-12:
Ah, thank you!

nnagururu, 2021-04-12:
Hmm, I have the same settings and I'm not getting color...

[image]

adnanmunawar, 2021-04-12:
Hmm, that is odd. Let's try a zoom session to figure out what's going on.

nnagururu, 2021-04-24:
I think it was just a problem in the parallels virtual machine as you suggested Adnan.

Color is working on the linux workstation I am now using. Thank you.

closed 2021-04-24 by nnagururu