WPI-AIM/ambf #129: transparent object

Zimmo95, 2021-04-12:
Hi Adnan, 
Is it possible to create and import a transparent object?

Enrico

adnanmunawar, 2021-04-12:
Hi Enrico, Yes, you can set the transparency value in the ADF for the object or use the material tab in the ambf_addon. This is a video of how to do that. Note, when you enable the material, make sure to click "use nodes" to toggle the node-based rendering (in Blender) to off. Then you can set the color/transparency using the color wheel. Now when you save the ADF, the data should be consistent in ADF.

[image]

Zimmo95, 2021-04-13:
Hi Adnan, 
It works perfectly.
Thanks for your help

Enrico

closed 2021-04-13 by adnanmunawar