WPI-AIM/ambf #130: virtual reality

Zimmo95, 2021-04-13:
Hi Adnan,
Is it possible to implement ambf on a virtual reality visor?
In particular, I am using HTC vive visor.

Enrico

adnanmunawar, 2021-04-14:
Hi Zimmo,

AMBF itself does not possess any special features to support or restrict using HMDs. In the past, we were able to use OpenHMD with Occulus Rift (camera orientation control but no position control). After configuring with OpenHMD, the VR HMD basically shows up as an external monitor in Linux. Then it's just a matter of enabling stereo on a specific ambf camera, and putting that simulation window over to the VR screen. Finally, you can use the Python client to control the camera pose using the pose values retrieved from the IMUs on the HMD. 

You can checkout OpenHMD project here:
https://github.com/OpenHMD/OpenHMD/wiki/Support-List

Zimmo95, 2021-04-16:
Hi Adnan,
I followed the instruction in the link and I succeeded to connect the device to my computer.
[picture]
** Then it's just a matter of enabling stereo on a specific ambf camera, and putting that simulation window over to the VR screen** How can I do this?

**Finally, you can use the Python client to control the camera pose using the pose values retrieved from the IMUs on the HMD.** It means that is possible to publish IMUs values on ambf even though my HTC vive is not connected to ROS?

Thanks for your time,
Enrico
[image]

adnanmunawar, 2021-04-21:
Take a look at the [world](https://github.com/WPI-AIM/ambf/blob/ambf-1.0/ambf_models/descriptions/world/world.yaml) file for instance. There is a camera called `cameraEndoscope` which is not being used because it is not included in the list of cameras at line number 3. Regardless, it has the field called `stereo` that can be added (and uncommented) to any camera's description to make it Left and Right stereo:

``` yaml
cameraEndoscope:
  namespace: cameras/
  name: endo_camera
  location: {x: 0.0, y: 0.0, z: -0.1}
  look at: {x: 0.0, y: 0.0, z: -0.5}
  up: {x: 1.0, y: 0.0, z: 0.0}
  clipping plane: {near: 0.01, far: 10.0}
  field view angle: 0.8
  monitor: 0
  # stereo: {mode: Passive, eye separation: 0.22, focal length: 2.0}
  parent: /ambf/env/ecm/BODY remote center link
  # multipass: True
```

>  It means that is possible to publish IMUs values on ambf even though my HTC vive is not connected to ROS?

Not really, you need to access the IMU values whichever way you can/want and then use the Python client / ROS topics from AMBF to get/set the pose of the appropriate camera. Here is an example that we used for Occulus Rift 2:

We used this code to publish the Occulus Rifts IMUs as ROS messages:
https://github.com/h3ct0r/openhmd_ros

and then this code to do something (orient the camera) with those IMU values:
https://github.com/WPI-AIM/ambf/blob/ambf-1.0/ambf_ros_modules/ambf_comm/scripts/tests/occulus_view.py

This code in the second link is stale now. You will have to change the lines here and all other occurrence of `ObjectState` and `ObjectCmd` to `CameraState` and `CameraCmd`. E.g.:
``` python
    occulus_sub = rospy.Subscriber("/ambf/env/default_camera/State", ObjectState, occulus_cb, queue_size=1)
    occulus_pub = rospy.Publisher("/ambf/env/default_camera/Command", ObjectCmd, queue_size=1)
```
to

``` python
    occulus_sub = rospy.Subscriber("/ambf/env/default_camera/State", CameraState, occulus_cb, queue_size=1)
    occulus_pub = rospy.Publisher("/ambf/env/default_camera/Command", CameraCmd, queue_size=1)
```
and don't forget to include the appropriate headers for these new camera messages.

Zimmo95, 2021-04-22:
Hi Adnan,
Thank you for your help!

Best regards,
Enrico

closed 2021-04-27 by adnanmunawar