WPI-AIM/ambf #131: Error with collision meshes

Zimmo95, 2021-04-28:
Hi Adnan,
I have some errors when I launch my scene on ambf:
INFO! LOADING LOW RES MESH NAMES ../../ambf_models/descriptions/./multi-bodies/aorta_xray/../../../meshes/aorta_xray/low_res/Aorta5.STL
ERROR! FOR Aorta5 COLLISION MESH TYPE NOT UNDERSTOOD
Thread Joined: Aorta5
[for each aorta#.stl]

This error is related to the drastic lowering of the frequency Phx below 5 Hz

My Aorta is made of 45 different segments with concave collision meshes.

Thanks for your help,
Enrico

adnanmunawar, 2021-04-29:
Hi Enrico,
Can you post the relevant portion of the ADF file.

Zimmo95, 2021-04-30:
BODY Aorta6:
  name: Aorta6
  mesh: Aorta6.STL
  mass: 0.0
  inertia:
    ix: 0.0
    iy: 0.0
    iz: 0.0
  collision margin: 0.001
  scale: 1.0
  location:
    position:
      x: 7.061
      y: -0.7734
      z: -0.0334
    orientation:
      r: -0.0627
      p: -0.1387
      y: -1.9487
  inertial offset:
    position:
      x: 1.5183
      y: -4.5002
      z: 1.9244
    orientation:
      r: 0
      p: 0
      y: 0
  passive: false
  publish children names: false
  publish joint names: false
  publish joint positions: false
  friction:
    rolling: 0.0
    static: 0.5
  restitution: 0.1
  damping:
    angular: 0.1
    linear: 0.04
  collision groups: []
  collision mesh type: CONCAVE_MESH
  color components:
    ambient:
      level: 1.0
    diffuse:
      b: 0.0348
      g: 0.0348
      r: 0.8
    specular:
      b: 0.0174
      g: 0.0174
      r: 0.4
    transparency: 1.0
BODY Aorta7:
  name: Aorta7
  mesh: Aorta7.STL
  mass: 0.0
  inertia:
    ix: 0.0
    iy: 0.0
    iz: 0.0
  collision margin: 0.001
  scale: 1.0
  location:
    position:
      x: 7.061
      y: -0.7734
      z: -0.0334
    orientation:
      r: -0.0627
      p: -0.1387
      y: -1.9487
  inertial offset:
    position:
      x: 1.3418
      y: -2.2889
      z: 0.8046
    orientation:
      r: 0
      p: 0
      y: 0
  passive: false
  publish children names: false
  publish joint names: false
  publish joint positions: false
  friction:
    rolling: 0.0
    static: 0.5
  restitution: 0.1
  damping:
    angular: 0.1
    linear: 0.04
  collision groups: []
  collision mesh type: CONCAVE_MESH
  color components:
    ambient:
      level: 1.0
    diffuse:
      b: 0.0348
      g: 0.0348
      r: 0.8
    specular:
      b: 0.0174
      g: 0.0174
      r: 0.4
    transparency: 1.0
BODY Aorta8:
  name: Aorta8
  mesh: Aorta8.STL
  mass: 0.0
  inertia:
    ix: 0.0
    iy: 0.0
    iz: 0.0
  collision margin: 0.001
  scale: 1.0
  location:
    position:
      x: 7.061
      y: -0.7734
      z: -0.0334
    orientation:
      r: -0.0627
      p: -0.1387
      y: -1.9487
  inertial offset:
    position:
      x: 1.9222
      y: -3.6142
      z: 1.1412
    orientation:
      r: 0
      p: 0
      y: 0
  passive: false
  publish children names: false
  publish joint names: false
  publish joint positions: false
  friction:
    rolling: 0.0
    static: 0.5
  restitution: 0.1
  damping:
    angular: 0.1
    linear: 0.04
  collision groups: []
  collision mesh type: CONCAVE_MESH
  color components:
    ambient:
      level: 1.0
    diffuse:
      b: 0.0348
      g: 0.0348
      r: 0.8
    specular:
      b: 0.0174
      g: 0.0174
      r: 0.4
    transparency: 1.0

adnanmunawar, 2021-05-03:
The error is related to Aorta5.stl which I don't see in the contents that you pasted above. Can you instead zip your ADF file and the meshes and share it.

Zimmo95, 2021-05-03:
Aorta5 exists. I pasted only a part of the file. 
In the terminal I have an error for every stl. 
Anyhow, the environment works correctly. 

Enrico

adnanmunawar, 2021-05-03:
> Aorta5 exists. I pasted only a part of the file.
In the terminal I have an error for every stl.

Oh, I see.

> Anyhow, the environment works correctly.

So is there issue no longer there?

Zimmo95, 2021-05-03:
Fortunately there is no issues but the error lines are still there. 

Thanks for your time, 
Enrico

Zimmo95, 2021-05-03:
Fortunately there are no issues but the error lines are still there. 

Thanks for your time,
Enrico

adnanmunawar, 2021-05-07:
Does this issue still persist with the restructure branch?

Zimmo95, 2021-05-07:
No, the problem disappeared. 

Thanks for your time,
Enrico

closed 2021-05-07 by Zimmo95