WPI-AIM/ambf #132: python Client

Zimmo95, 2021-05-06:
Hi Adnan,
I cannot get the position (and orientation) of the objects in my environment.
After launching the python application and following your instructions, the position of every object is:
x: 0.0
y: 0.0
z: 0.0
 even if I move them with the "set_pose" line 

Best,
Enrico

adnanmunawar, 2021-05-06:
That is strange, never seen this before. I am going to need more info though, a sample of the Python script that you are running, the ADF file, and the AMBF branch and commit that you are on.

Zimmo95, 2021-05-06:
[Aorta.zip](https://github.com/WPI-AIM/ambf/files/6436130/Aorta.zip)
[python_script.zip](https://github.com/WPI-AIM/ambf/files/6436159/python_script.zip)
What do you mean with "AMBF branch and commit that you are on"?

adnanmunawar, 2021-05-06:
You can browse to the location where you cloned AMBF repo and then run `git branch -v` and paste the output, and then `git log --oneline` and also paste that output.

adnanmunawar, 2021-05-06:
Btw, I just briefly looked at your Python script. Before I try anything, can you put a sleep method in your code and then confirm if the issue persists. Here is roughly what it should look like.

``` python
import time
from ambf_client import Client
c = Client()
c.connect()
time.sleep(0.2)
t = c.get_obj_handle('/ambf/env/Sphere')
time.sleep(0.2)
pos = t.get_pos()
print(pos)
```

Zimmo95, 2021-05-06:
* ambf-1.0 e1206fa5 Updated examples with images

e1206fa5 (HEAD -> ambf-1.0, origin/ambf-1.0, origin/HEAD) Updated examples with images
3d6ddb7b Added an example of teleop with input device
4e4402c9 Update the base and tip transforms in proxy_device.py
9e658b38 Added example of input_device_mapping
651fd1e1 Fix the setting of initial rot if location not specified and workspace scaling
1f41f5eb Raising the min allowed value of dt for joint controller
0cc50699 Bug, replacing == with = for setting joint controller output type.
af736a55 Refactoring workspace scaling term and adding damping ramp of orientation control.
1d891b05 Bug, fixed class import
a639ee5b Bug, replacing == by = in controller output mode type assignment.
fa683f77 Merge remote-tracking branch 'origin/ambf-1.0' into ambf-1.0
cdf9dcfb Fixed softbody orientation
19c4a02a Adding the capability of creating convex hulls and convex triangle based collision from meshes.
d657f7b0 AMBF C++ client (#106)
8909b9db Fixed typo in readme
4c910cf5 Fix #92. Added example of sensors and actuators and grasping based on them.
0227bd3e Removed explicit joint controllers for the KUKA so that velocity based controllers are used
:

I tried to launch your script but the error still exists

adnanmunawar, 2021-05-06:
Great, I would recommend switching to the `restructure` branch and the newest commit, although it doesn't have anything to do with this issue, it might be related to the other issue here https://github.com/WPI-AIM/ambf/issues/131. 

Now I ran the ADF file with the exact branch and commit and this is the output that I get:

``` bash
python ~/Downloads/python_script.py 
('Found Common Object Namespace as: ', '/ambf/env/')
x: 1.1883
y: -0.2964
z: 2.2166
('Closing publisher for: ', '/ambf/env/Aorta45')
('Closing publisher for: ', '/ambf/env/Plane1')
('Closing publisher for: ', '/ambf/env/Aorta42')
...
```
Can you try increasing the sleep time to a couple of seconds and see what you get?

Zimmo95, 2021-05-06:
I increased to 2 seconds but the problem still exists.
How can I switch to restructure branch and newest commit
?

adnanmunawar, 2021-05-06:
I would recommend getting familiar with git using a plethora of online resources. Here is an easy one https://git-scm.com/book/en/v2/Git-Branching-Basic-Branching-and-Merging

Zimmo95, 2021-05-07:
I deleted and re-downloaded ambf and now everything works..LOL

adnanmunawar, 2021-05-07:
Great. 
Please remember to close the issues once they are resolved.

closed 2021-05-07 by Zimmo95