WPI-AIM/ambf #133: Collision meshes

vanessaloia, 2021-05-17:
Hello, 

I would like to do a simple pick & place task using the `puzzle6.yaml` and two PSMs (`psm1.yaml` and `psm2.yaml`) . 
When the gripper interacts with the objects in the scene, it passes through them so I'm not able to pick the triangles even if the collision meshes were set properly as I could see from Blender. 

Any suggestion? 
Thanks

adnanmunawar, 2021-05-17:
Did you create the `psm1.yaml` and `psm2.yaml` files yourself? 
AMBF allows for multi-level collision filtering and here is a quick explanation for it.
For a specific ADF file, `A.yaml`, there is a global boolean flag called `ignore inter-collision`. If set to `true`, no objects described in the `A.yaml` file would collide with each other. This flag only applies to the ADF file in question and has no effect on objects from another ADF file from colliding with objects in `A.yaml`.

For finer control of collision filtering, each object (rigid body) has an optional array field called `collision groups`. This array can be allocated with positive integer indices. All the objects in the scene (defined using a single or multiple ADF files) that have a common collision group index will collide with each other, and the ones that do not have a common collision group index will not collide with each other. If the `collision groups` array for an object is empty, it means it has no collision filtering and will collide with all objects in the scene (and with objects in the scene defined via the same ADF file if `ignore inter-collision: False`).

Of course the collision groups can be set to have more advanced collision filtering, here is an example.
``` yaml
body1:
collision groups: [0]

body2:
collision groups: [0, 1]

body3:
collision groups: [1, 2]

body4:
collision groups: [2]
```

In the minimal example above, the following collision filtering ensues:

**COLLISION ENABLED**:
```

body1 with body2 (both contain collision group `0`)
body2 and body3 (both contain collision group `1`)
body3 and body4 (body contain collision group `2`)
```

**COLLISION NOT ENABLED**:
```
body1 with body3 (no common collision group)
body1 with body4 (no common collision group)
body2 with body4 (no common collision group) 
```
You can set both the `ignore inter-collision` flag and `collision groups` via the `ambf_addon` or manually edit the ADF file(s).

vanessaloia, 2021-05-18:
Hello, 

thanks for the explanation. 
As you suggested, I cluestered the grippers and the triangles (the shapes that have to be picked) into the same collision group and enable the ignore inter-collision to true for all the ADF files. 
Right now, the grippers  collide with the triangles but I'm still not able to performe a pick & place task since the objects of the puzzles don't collide even if the inter-collision flag is set to true. 
I'm attacching the ADF of the puzzle for a better understand, maybe I'm missing something. The psm1.yaml and psm2.yaml are the one I found in the surgical challenge. 

Moreover do you know why the models created in Blender have a different representation (for example position or orinetation) once launched in the simulation? 

[models.zip](https://github.com/WPI-AIM/ambf/files/6501104/models.zip)

thanks

adnanmunawar, 2021-05-19:
> The psm1.yaml and psm2.yaml are the one I found in the surgical challenge.

Oh I see, great.

> Right now, the grippers collide with the triangles but I'm still not able to performe a pick & place task since the objects of the puzzles don't collide even if the inter-collision flag is set to true.

The `ignore inter-collision` flag should be set to `False` if you want the objects in that ADF file to collide with each other. 

 > Moreover do you know why the models created in Blender have a different representation (for example position or orinetation) once launched in the simulation?

They should not have a different position or orientation compared. Why do you think that there position and orientation is different once launched in the simulator?