WPI-AIM/ambf #135: Shaders Not Loading

nnagururu, 2021-05-21:
Preprocessing shaders not loading properly, see initialization message below:

```
INFO! WORLD CONFIG PATH: ../../ambf_models/descriptions/./world 
Thread Joined: World
INFO! SETTING SIMULATION MAX ITERATIONS TO : 50
Thread Joined: light2
shader compilation failed:
0:14(7): error: syntax error, unexpected NEW_IDENTIFIER, expecting ',' or ';'

shader compilation failed:
0:14(7): error: syntax error, unexpected NEW_IDENTIFIER, expecting ',' or ';'
Thread Joined: default_camera
shader compilation failed:
0:14(7): error: syntax error, unexpected NEW_IDENTIFIER, expecting ',' or ';'

shader compilation failed:
0:14(7): error: syntax error, unexpected NEW_IDENTIFIER, expecting ',' or ';'
Thread Joined: segm_camera
INFO! LOADING LOW RES MESH NAMES ../../ambf_models/descriptions/./mastoidectomy/../../meshes/mastoidectomy/low_res/Drill.STL
INFO! NUMBER OF MESHES FOR CONVEX HULL 1
CREATING CONVEX HULL FROM MESH NUMBER 0
INFO! FOR Drill CREATING CONVEX HULL FROM COLLISION MESH (CASE 1)
Thread Joined: Drill
INFO! LOADING LOW RES MESH NAMES ../../ambf_models/descriptions/./mastoidectomy/../../meshes/mastoidectomy/low_res/Mastoid.STL
INFO! NUMBER OF MESHES FOR CONVEX HULL 1
CREATING CONVEX HULL FROM MESH NUMBER 0
INFO! FOR Mastoid CREATING CONVEX HULL FROM COLLISION MESH (CASE 1)
Thread Joined: Mastoid
Found Lib: libdvrk_arm.so
No of dVRK Masters detected: 0
WARNING: FAILED TO LOAD DEVICE: "MTMR"
WARNING: FAILED TO LOAD DEVICE: "MTML"
WARNING: FAILED TO LOAD DEVICE: "Geomagic Touch"
WARNING: FAILED TO LOAD DEVICE: "Razer Hydra0"
WARNING: FAILED TO LOAD DEVICE: "Razer Hydra1"
WARNING: FAILED TO LOAD DEVICE: "Novint Falcon1"
WARNING: FAILED TO LOAD DEVICE: "Novint Falcon2"
```

Also attached is an image of my computer specs:
[image]

adnanmunawar, 2021-05-27:
I think this issue is the same as Issue #127. The commit above should address it nonetheless.

nnagururu, 2021-05-30:
That fixed it! Thanks

closed 2021-05-30 by nnagururu