WPI-AIM/ambf #140: get_pos() doesn't work

Zimmo95, 2021-06-29:
Hi Adnan,
I tried to move a robot in predefined positions using get_pos and everything worked.
When I tried to do the same task connected to ROS master via ethernet the robot stopped to move even though the positions were updated correctly. 
I failed also to do the same task using the phantom omni: the phantom published but the robot didn't move (these python client codes the last month worked perfectly)

Enrico

adnanmunawar, 2021-07-02:
Hi Enrico,
I don't think I understand the issue as you described. Can you post a small code snippet of what you are trying to do?

enricokarim, 2021-07-02:
I got the same problem of issue #132. 
I suppose that there is some problem with the connection to Ros master in my lab. 
Finally, I re-downloaded AMBF and everything restarted to work. 

Thanks for you time. 
Enrico

Zimmo95, 2021-07-02:
I got the same problem of issue #132.
I suppose that there is some problem with the connection to Ros master in my lab.
Finally, I re-downloaded AMBF and everything restarted to work.

Thanks for you time.
Enrico

closed 2021-07-02 by Zimmo95