WPI-AIM/ambf #144: Build Error - '../bin/lin-x86_64/ambf_simulator' failed

francomomo, 2021-11-01:
Hi Adnan,

I am working on a project in PAIR lab at the University of Toronto. I am struggling to finish the build process.

I am trying this build process on a Ubuntu 18.04 container in docker on my MacBook with Apple Silicon M1.

I am getting this error when running the make command. I am wondering if you can provide some insight on what might go wrong with this.

```
Scanning dependencies of target ambf_simulator
[ 93%] Building CXX object ambf_simulator/CMakeFiles/ambf_simulator.dir/src/ambf_simulator.cpp.o
[ 94%] Linking CXX executable ../../bin/lin-x86_64/ambf_simulator
../libambf_framework.a(CDeltaDevices.cpp.o): In function `chai3d::cDeltaDevice::close()':
CDeltaDevices.cpp:(.text+0x54): undefined reference to `drdClose'
CDeltaDevices.cpp:(.text+0xa4): undefined reference to `dhdClose'
../libambf_framework.a(CDeltaDevices.cpp.o): In function `chai3d::cDeltaDevice::getJointAnglesRad(double*)':
CDeltaDevices.cpp:(.text+0xe8): undefined reference to `dhdEnableExpertMode'
CDeltaDevices.cpp:(.text+0xf4): undefined reference to `dhdGetJointAngles'
CDeltaDevices.cpp:(.text+0xfc): undefined reference to `dhdDisableExpertMode'
../libambf_framework.a(CDeltaDevices.cpp.o): In function `chai3d::cDeltaDevice::getUserSwitches(unsigned int&)':
CDeltaDevices.cpp:(.text+0x17c): undefined reference to `dhdGetSystemType'
CDeltaDevices.cpp:(.text+0x1b8): undefined reference to `dhdGetButtonMask'
CDeltaDevices.cpp:(.text+0x1cc): undefined reference to `dhdGetButton'
../libambf_framework.a(CDeltaDevices.cpp.o): In function `chai3d::cDeltaDevice::getLinearVelocity(chai3d::cVector3d&)':
CDeltaDevices.cpp:(.text+0x250): undefined reference to `dhdGetLinearVelocity'
../libambf_framework.a(CDeltaDevices.cpp.o): In function `chai3d::cDeltaDevice::getPosition(chai3d::cVector3d&)':
CDeltaDevices.cpp:(.text+0x318): undefined reference to `dhdGetPosition'
../libambf_framework.a(CDeltaDevices.cpp.o): In function `chai3d::cDeltaDevice::getGripperAngleRad(double&)':
CDeltaDevices.cpp:(.text+0x40c): undefined reference to `dhdGetGripperAngleRad'
../libambf_framework.a(CDeltaDevices.cpp.o): In function `chai3d::cDeltaDevice::getRotation(chai3d::cMatrix3d&)':
CDeltaDevices.cpp:(.text+0x5f4): undefined reference to `dhdGetOrientationRad'
CDeltaDevices.cpp:(.text+0x894): undefined reference to `dhdGetOrientationFrame'
../libambf_framework.a(CDeltaDevices.cpp.o): In function `chai3d::cDeltaDevice::open()':
CDeltaDevices.cpp:(.text+0x930): undefined reference to `drdOpenID'
CDeltaDevices.cpp:(.text+0xb60): undefined reference to `dhdOpenID'
CDeltaDevices.cpp:(.text+0x153c): undefined reference to `dhdEnableExpertMode'
CDeltaDevices.cpp:(.text+0x157c): undefined reference to `dhdSetWatchdog'
CDeltaDevices.cpp:(.text+0x1588): undefined reference to `dhdIsLeftHanded'
CDeltaDevices.cpp:(.text+0x1594): undefined reference to `dhdGetSystemType'
CDeltaDevices.cpp:(.text+0x15a4): undefined reference to `dhdSetVelocityThreshold'
../libambf_framework.a(CDeltaDevices.cpp.o): In function `chai3d::cDeltaDevice::getNumDevices()':
CDeltaDevices.cpp:(.text+0x1954): undefined reference to `dhdGetDeviceCount'
../libambf_framework.a(CDeltaDevices.cpp.o): In function `chai3d::cDeltaDevice::cDeltaDevice(unsigned int, bool)':
CDeltaDevices.cpp:(.text+0x1adc): undefined reference to `dhdGetDeviceCount'
../libambf_framework.a(CDeltaDevices.cpp.o): In function `chai3d::cDeltaDevice::enableForces(bool)':
CDeltaDevices.cpp:(.text+0x1c2c): undefined reference to `dhdEnableExpertMode'
CDeltaDevices.cpp:(.text+0x1c38): undefined reference to `dhdEnableForce'
CDeltaDevices.cpp:(.text+0x1c40): undefined reference to `dhdDisableExpertMode'
CDeltaDevices.cpp:(.text+0x1c64): undefined reference to `dhdEnableExpertMode'
CDeltaDevices.cpp:(.text+0x1c74): undefined reference to `dhdEnableForce'
CDeltaDevices.cpp:(.text+0x1c7c): undefined reference to `dhdDisableExpertMode'
../libambf_framework.a(CDeltaDevices.cpp.o): In function `chai3d::cDeltaDevice::calibrate(bool)':
CDeltaDevices.cpp:(.text+0x1d34): undefined reference to `drdIsInitialized'
CDeltaDevices.cpp:(.text+0x1d48): undefined reference to `drdAutoInit'
CDeltaDevices.cpp:(.text+0x1d58): undefined reference to `drdStop'
CDeltaDevices.cpp:(.text+0x1d64): undefined reference to `drdIsInitialized'
CDeltaDevices.cpp:(.text+0x1d80): undefined reference to `dhdReset'
../libambf_framework.a(CDeltaDevices.cpp.o): In function `chai3d::cDeltaDevice::setForceAndTorqueAndGripperForce(chai3d::cVector3d const&, chai3d::cVector3d const&, double)':
CDeltaDevices.cpp:(.text+0x1fa4): undefined reference to `dhdSetForceAndGripperForce'
CDeltaDevices.cpp:(.text+0x1ff0): undefined reference to `dhdSetForceAndTorqueAndGripperForce'
CDeltaDevices.cpp:(.text+0x2040): undefined reference to `dhdGetGripperAngleRad'
CDeltaDevices.cpp:(.text+0x20ac): undefined reference to `dhdSetForce'
../libambf_framework.a(CDeltaDevices.cpp.o): In function `chai3d::cDeltaDevice::setDeviceAngleDeg(double)':
CDeltaDevices.cpp:(.text+0x2168): undefined reference to `dhdSetDeviceAngleDeg'
../libambf_framework.a(CDeltaDevices.cpp.o): In function `chai3d::cDeltaDevice::setVibration(double, double, int)':
CDeltaDevices.cpp:(.text+0x21b4): undefined reference to `dhdSetVibration'
collect2: error: ld returned 1 exit status
ambf_simulator/CMakeFiles/ambf_simulator.dir/build.make:250: recipe for target '../bin/lin-x86_64/ambf_simulator' failed
make[2]: *** [../bin/lin-x86_64/ambf_simulator] Error 1
CMakeFiles/Makefile2:3624: recipe for target 'ambf_simulator/CMakeFiles/ambf_simulator.dir/all' failed
make[1]: *** [ambf_simulator/CMakeFiles/ambf_simulator.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
```

adnanmunawar, 2021-11-01:
Hi Ka Chun,

Thanks for the introduction and for opening this issue. Let's see if we can get to the bottom of it. 

Can you go to this file [cGlobals.h](https://github.com/WPI-AIM/ambf/blob/ambf-2.0/external/chai3d/src/system/CGlobals.h) and comment out the [line 215](https://github.com/WPI-AIM/ambf/blob/ambf-2.0/external/chai3d/src/system/CGlobals.h#L215):

so that it looks like this

```cpp
    #if defined(LINUX)

    //--------------------------------------------------------------------
    // GENERAL
    //--------------------------------------------------------------------

    // OS specific
    #include <ctime>
    #include <pthread.h>
    #include <dlfcn.h>

    // printf
    #define cPrint printf

    //--------------------------------------------------------------------
    // HAPTIC DEVICES
    //--------------------------------------------------------------------
    #if !defined (__arm__)
    //    #define C_ENABLE_DELTA_DEVICE_SUPPORT
    #endif
    #define C_ENABLE_PHANTOM_DEVICE_SUPPORT
    #define C_ENABLE_LEAP_DEVICE_SUPPORT
//    #define C_ENABLE_AMBF_DVRK_DEVICE_SUPPORT
//    #define C_ENABLE_SIXENSE_DEVICE_SUPPORT

#endif
```

and then try to rebuild.

francomomo, 2021-11-02:
Hello Adnan,

I am able to build successfully now!! But now I am seeing the issue that `ERROR! FAILED TO INITIALIZE GLFW LIBRARY`

`root@cbdc51992c51:~/ambf/bin/lin-x86_64# ./ambf_simulator 

____________________________________________________________

ASYNCHRONOUS MULTI-BODY FRAMEWORK SIMULATOR (AMBF Simulator)

                (www.aimlab.wpi.edu)
                  (Copyright 2019-2021)
____________________________________________________________

STARTUP COMMAND LINE OPTIONS: 

ambf_simulator Command Line Options:
  -h [ --help ]                        Show help
  -n [ --ndevs ] arg (=0)              Number of Haptic Devices to Load
  -i [ --load_devices ] arg            Index number of devices to load which is
                                       specified in input_device.yaml
  -e [ --enableforces ] arg (=0)       Enable Force Feedback on Haptic Devices
  -p [ --phx_frequency ] arg (=1000)   Physics Update Frequency (default: 1000 
                                       Hz)
  -d [ --htx_frequency ] arg (=1000)   Haptics Update Frequency (default: 1000 
                                       Hz)
  -t [ --fixed_phx_timestep ] arg (=0) Use Fixed Time-Step for Physics 
                                       (default: False)
  -f [ --fixed_htx_timestep ] arg (=0) Use Fixed Time-Step for Haptics 
                                       (default: False)
  -g [ --show_gui ] arg (=1)           Show GUI
  --ns arg                             Override the default (or specified in 
                                       ADF) world namespace
  -s [ --sim_speed_factor ] arg (=1)   Override the speed of "NON REAL-TIME" 
                                       simulation by a specified factor 
                                       (Default 1.0)
  - [ --plugins ] arg                  Simulator plugins to load, .e.g. 
                                       --plugins <plugin1_filepath>, 
                                       <plugin2_filepath> loads plugin1 and 
                                       plugin2 simualtor plugin
  --launch_file arg                    Launch file path to load (default: 
                                       <ROOT_PATH>/ambf_models/descriptions/lau
                                       nch.yaml
  -a [ --load_multibody_files ] arg    Description Filenames of Multi-Body(ies)
                                       to Launch, .e.g. -a <path>/test.yaml, 
                                       <another_path>/test2.yaml will load 
                                       multibodies test.yaml and test2.yaml if 
                                       they are valid files
  -l [ --load_multibodies ] arg        Index of Multi-Body(ies) to Launch, 
                                       .e.g. -l 1,2,3 will load multibodies at 
                                       indexes 1,2,3. See launch.yaml file

------------------------------------------------------------

ERROR! FAILED TO INITIALIZE GLFW LIBRARY`

adnanmunawar, 2021-11-02:
Can you tell me which base docker image are you using? I have had success with using images from this repo, as they have Ubuntu + ROS + VNC. https://github.com/henry2423/docker-ros-vnc

francomomo, 2021-11-03:
I am just using the regular ubuntu:18.04 image. I'll try out the images you mentioned!!

francomomo, 2021-11-03:
Do you know if there are any ways to have a visualization of the ambf simulator when running it in a docker environment on MacBook m1? Since we are running it in a container I do not know any way to have a visualization of the simulator. I would appreciate some guidance!!

adnanmunawar, 2021-11-04:
If you check out the Readme of the docker image that I posted above (https://github.com/henry2423/docker-ros-vnc), they have listed the instructions on running the visualization using VNC. Once you have the visualization of the Linux OS installed by the Docker image, you can run graphical applications, like AMBF, as you would normally run them on a native OS.

francomomo, 2021-11-11:
Hi! I ran into this `qemu:handle_cpu_signal received signal outside vCPU context @ pc=0x356d52` issue when i ran `./ambf_simulator`. I am not sure if this is related to docker or running docker on MacBook M1. Have you guys seen similar issues in the past?
[image]
The window above is what shows up when I ran `./ambf_simulator`. The window below is when I ran `roscore`

adnanmunawar, 2021-11-12:
This is the first time that I am seeing this issue. It may be an issue relating to the combination of docker, M1 chip, and ambf. Does roscore give that error message in red even before running AMBF? Can you try running other GUI apps in the container, such a Gedit, and Gazebo?

francomomo, 2021-11-14:
Hi Adnan! Thank you for your support along the way. I am able to run `ambf_simulator` in Ubuntu 20.04 on parallels on Macbook M1!

closed 2021-11-14 by francomomo