WPI-AIM/ambf #149: Softbody-Rigidbody collisions

JohannesSchwaiger, 2021-12-01:
Hi Adnan,

I am currently trying out the softbody capabilities and want a rigid body to interact with a king of rubber band. But any sort of collision or sometimes even near collisions create chaotic behaviour in the rubber band. 
Using the mouse to manipulate the rubber band works fine.
I tried the cheese example with better success, so I am assuming my softbody parameters are incorrect. I tried your parameters from this issue https://github.com/WPI-AIM/ambf/issues/39 but to no avail. 
My file looks like this, 
`# AMBF Version: (0.1)
# Generated By: ambf_addon for Blender (2.79.0)
# Link: https://github.com/WPI-AIM/ambf_addon
# Generated on: 2019-06-23 03:33:54
soft bodies: [BODY Cloth]
bodies:
joints: []
high resolution path: ./low_res/
low resolution path: ./low_res/
ignore inter-collision: true
namespace: /ambf/env/
BODY Cloth:
  name: cloth
  mesh: straight_small.obj
  mass: 1.0
  collision margin: 0.001
  scale: 1.0
  location:
    orientation: {p: -0.0, r: 0.0, y: 0.0}
    position: {x: -1, y: 0.0, z: -1.0}
  inertial offset:
    orientation: {p: 0, r: 00, y: 0}
    position: {x: 0.0, y: 0.0, z: 0.0}
  color components:
    ambient: {level: 1.0}
    diffuse: {b: 0.8, g: 0.8, r: 0.8}
    specular: {b: 1.0, g: 1.0, r: 1.0}
    transparency: 1.0
  config: {
    kLST: 0.005,
    # kAST: 0.5,
    kDP: 0.001,
    # kDF: 0.5,
    # kMT: 0.01,
    piterations: 6,
    bending constraint: 2,
    margin: 0.01,
    # kPR: 2,
    # kVC: 10,
    # kVST: 0.1,
    # kAST: 1.0,
    flags: 18,
    clusters: 0,
    fixed nodes: [0,160,8,161,4,5,2,11]
    }
  randomize constraints: true
  collision margin: 0.05
  color components:
    ambient: {level: 1.0}
    diffuse: {b: 0.0, g: 0.0, r: 0.0}
    specular: {b: 1.0, g: 1.0, r: 1.0}
    transparency: 1.0`

My rubber band looks like this in blender: 
[image]

Trying contact with meshes for collision drops my physics-framerate to 1-10 even just getting close. Using primitives I at least can get close, but contact still results in chaotic behaviour, especially of the softbody.

How can I achieve stable interaction between rigid bodies and something like a rubber band?

Sidenote: Also indices of vertices in blender don't seems to correspond to the indices specified in fixed nodes. How do I know which indices are which so i can fix both sides?

Sorry for the long question and thank you for your work and help!

adnanmunawar, 2021-12-08:
Hello Johannes, Thank you for creating this issue. Can you please attach a zipped folder with your ADF file and the meshes in it so I can test it? Also, if you are using a custom rigid body, can you please also include its ADF files, or let me know which one from ambf_models are you using.

Here are the answers to specific questions:

> Trying contact with meshes for collision drops my physics-framerate to 1-10 even just getting close. Using primitives I at least can get close, but contact still results in chaotic behaviour, especially of the softbody.

>How can I achieve stable interaction between rigid bodies and something like a rubber band?

Your observation is correct. For softbodies, we would always want to collide with rigid bodies that are made up of collision primitives, unless the bodies are static, or have a very low vertex count. If you are already using primitives, we might not just tune some parameters.

> Sidenote: Also indices of vertices in blender don't seems to correspond to the indices specified in fixed nodes. How do I know which indices are which so i can fix both sides?

Yes, your observation is correct. This is something that we need to fix.

In general, there is some work needed on the soft body interface, we have done a lot of restructuring this year and we hope to get to the better soft body integration soon.

JohannesSchwaiger, 2021-12-08:
Hello Adnan,

thank you for your response!

Here you can find my [models](https://github.com/WPI-AIM/ambf/files/7678058/models.zip). One stick using primitives and a rubber band fixed on both sides. But I also used many different rigid bodies. 
For context: My goal is to create a rubber band fixed on both sides which then can be stretched out and held back by a stick on the side. Kind of like in this beatiful drawing. During the process i want to know the forces the rubber band produces on the interacting bodies. 

[image]

Many Thanks in advance

adnanmunawar, 2021-12-14:
Hi Johannes,

Thanks for explaining your intent, I like your drawing :D. I got a chance to test the model files and see what is going on. Here are a couple of things that I noticed and would be worth noting

1. The dimensions were too small. It is better to make objects larger, both for rigid bodies and especially for soft-bodies. 
2. The structure of the softbody, especially the sides of have a lot of vertices very near to each other (shown in the Fig below). This could cause issues. Reducing the vertex count on side faces or increasing the dimension axis (to increase the relative vertex distance) and probably tweaking the "margin" param may help.

[image]

I updated the models based on the two suggestions above and you can find it here:

[models_updated.zip](https://github.com/WPI-AIM/ambf/files/7714683/models_updated.zip)

Here a GIF of the result that I got:

[image]

Finally, we have our work cut out to improve softbody support, but hopefully this will help you somewhat.

adnanmunawar, 2021-12-15:
I got some time to try and fix the setting of "fixes nodes" using the vertex indices from the original mesh. You can try pulling 529ab34 and building it. I also created a small GIF on how to find the vertex indices using Blender, which should make the workflow easier:

[image]

For the example of the mesh above, the vertices at the four corners seem to be 

``` yaml
fixed nodes: [0,2,3,9,6,152,153,155]
```

JohannesSchwaiger, 2021-12-27:
Hi Adnan,

thanks a lot for your help! 
Is there a reason small scales dont work as well? I would like to use a small gripper to bend the rubber band behind the stick. But I guess I could scale up everything and the scale the forces acordingly.

Also thanks for the quick fix on the indices!

adnanmunawar, 2022-01-08:
Hi Johannes, 
There are several reasons for this, smaller scales result in less precision for everything. I would have to dig deeper to figure out exactly what is the critical component that causes instability in this specific case but it's always good to keep things greater than 0.1 dimensions and scale your outputs (forces in your case).

closed 2022-02-28 by JohannesSchwaiger