WPI-AIM/ambf #153: No cpuid.h file

mikami520, 2022-01-04:
Hi, when I install ambf-2.0 on Ubuntu 20.04, it gave an error that the "cpuid.h" is missing. Could you please help me solve it?
[image]

adnanmunawar, 2022-01-04:
What is the architecture of your system? Not sure if this is related, but can you try the suggestions from this issue https://github.com/WPI-AIM/ambf/issues/142.

mikami520, 2022-01-04:
> What is the architecture of your system? Not sure if this is related, but can you try the suggestions from this issue #142.

My architecture is ARM64, I use the VM on the M1 Macbook

adnanmunawar, 2022-01-04:
Thanks for the info. I don't have access to an M1 Mac and thus haven't tried to build AMBF on a VM. Did you try the solution to Issue #142 and did it change anything? 

If that doesn't work for you, I know of a couple of people who tried running AMBF via Docker on the M1 Mac. You can find the instructions for that here:

https://github.com/collaborative-robotics/docker-ambf

mikami520, 2022-01-04:
> Thanks for the info. I don't have access to an M1 Mac and thus haven't tried to build AMBF on a VM. Did you try the solution to Issue #142 and did it change anything?
>
> If that doesn't work for you, [...]

I will have a try and let you know the further result, thanks for your help!

adnanmunawar, 2022-01-05:
Great, could you also try this on your VM. In your local AMBF copy, can you comment out this line: https://github.com/WPI-AIM/ambf/blob/ambf-2.0/external/openal/include/alConfig.h#L74

in the file `ambf/external/openal/include/alConfig.h`
from 
```
#define HAVE_CPUID_H
```

```
//#define HAVE_CPUID_H
```
and see if this allows you to compile.

radianag, 2022-01-24:
@adnanmunawar I tested this workaround on my M1 mac mini running parallels VM Ubuntu 20.04 and it works.

adnanmunawar, 2022-01-24:
Thanks, Radian, the new commit should incorporate these changes.

youpmac, 2022-04-14:
> Thanks, Radian, the new commit should incorporate these changes.

@adnanmunawar hi Adnan, this commit did not prevent the error for me. However, commenting out those lines again did work. I guess the commit implementing the if statement does not always work. Just wanted to let you know. I'm using a M1 macbook air running Ubuntu 18.04 on VMWare Fusion.

adnanmunawar, 2022-04-15:
Thanks for reporting this Youp, I shall take a look at it. I might need your help to test it though as I do not have an M1 Mac :)

closed 2026-09-29 by YuliangXiaoYLX